euslisp/jskeus #161: equal でsegfaultする例がある

k-okada, 2015-01-11:
```
(setq a (instance propertied-object))
(send a :put :joint-coords (make-coords))
(setq b (instance propertied-object))
(send b :put :joint-coords (make-coords))
(print a)
(print (eq a b))
(print (equal a b))

(setq bs (list (make-cube 10 10 10)))
(setq a (instance bodyset-link :init (make-cascoords) :bodies bs :name :a))
(send a :put :joint-coords (make-cascoords))
(setq bs (list (make-cube 10 10 10)))
(setq b (instance bodyset-link :init (make-cascoords) :bodies bs :name :b))
;(send b :put :joint-coords (make-cascoords)) ;; dame                                                                  
;(send b :put :joint-coords (make-cascoords :pos (float-vector 1 2 3))) ;; ok                                          
(send b :put :joint-coords (make-cascoords :pos (float-vector 0 0 0))) ;; dame                                         
(print a)
(print (eq a b))
(print (equal a b))

```
でsegfaultします．書き方が悪い？それともなんかバグが有る？

gdbかけるとhttps://github.com/euslisp/EusLisp/blob/master/lisp/c/predicates.c#L175
で死んでいるようです．

garaemon, 2015-01-11:
バグがあるっぽいですね

garaemon, 2015-01-11:
僕のgdbだと
```
Program received signal SIGSEGV, Segmentation fault.
equal (x=0x47a27c8, y=0x47a27c8) at predicates.c:164
164     { register int i,n,xe;
(gdb) where
#0  equal (x=0x47a27c8, y=0x47a27c8) at predicates.c:164
#1  0x0000000000429f34 in equal (x=0x4ba3978, y=0x4bb8408)
    at predicates.c:177
#2  0x0000000000429f34 in equal (x=0x4902580, y=0x4a753f0)
    at predicates.c:177
#3  0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177
#4  0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177
#5  0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177
#6  0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177
#7  0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177
#8  0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177
#9  0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177
#10 0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177
#11 0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177
#12 0x0000000000429f34 in equal (x=0x4bb7e20, y=0x4bcc528)
    at predicates.c:177

```

garaemon, 2015-01-11:
https://github.com/euslisp/EusLisp/blob/master/lisp/c/predicates.c#L177
ここで無限ループになっている

garaemon, 2015-01-11:
cascoordsのmanagerが循環参照になっているからでしょうか?

garaemon, 2015-01-11:
```
13.irteusgl$ setq a (make-cascoords)
14.irteusgl$ a
#<cascaded-coords #X57a3398  0.0 0.0 0.0 / 0.0 0.0 0.0>
15.irteusgl$ (a . manager)
#<cascaded-coords #X57a3398  0.0 0.0 0.0 / 0.0 0.0 0.0>
```