euslisp/jskeus #164: catkin build fails

garaemon, 2015-01-26:
* `catkin build jskeus` works
* `catkin build` does not work
```
 [jskeus] ==> '/home/leus/ros/hydro/build/jskeus/build_env.sh /usr/bin/make install' in '/home/leus/ros/hydro/build/jskeus'
Install the project...
-- Install configuration: ""
CMake Error at cmake_install.cmake:38 (message):
  Scanning dependencies of target compile_jskeus

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  nil

  #<package #X5ed3078 USER>

  nil

  nil

  "/home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//Linux64/obj/"

  "/home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//Linux64/obj/"

  compiling file: irtmath.l

  gcc -g -c -o
[... 613 lines cut ...]
  /home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//Linux64/obj/irtbvh.o:

  g++: :
  /home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//Linux64/obj/irtcollada.o:

  g++: :
  /home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//Linux64/obj/irtpointcloud.o:

  make[5]: ***
  [/home/leus/ros/hydro/devel/share/euslisp/jskeus/eus//Linux64/lib/libirteusg.so]
   1

  make[4]: *** [CMakeFiles/compile_jskeus]  2

  make[3]: *** [CMakeFiles/compile_jskeus.dir/all]  2

  make[2]: *** [CMakeFiles/compile_jskeus.dir/rule]  2

  make[1]: *** [compile_jskeus]  2

make: *** [install]  1
[jskeus] <== '/home/leus/ros/hydro/build/jskeus/build_env.sh /usr/bin/make install' failed with return code '2'
kjk

```

garaemon, 2015-01-26:
https://github.com/jsk-ros-pkg/jsk_demos/pull/141
This might be same issue.

@k-okada Could you take over this issue?

k-okada, 2015-01-26:
> jsk-ros-pkg/jsk_demos#141

yes

garaemon, 2015-01-26:
I found that unset DISPLAY works.
```
$ export DISPLAY=
$ catkin build
```

k-okada, 2015-01-26:
humm, I can not reproduce this....
https://gist.github.com/k-okada/4b9b9cb2886368d66c30

garaemon, 2015-01-26:
The environment now I'm testing sets DISPLAY=0:0, it may be a problem...

garaemon, 2015-01-26:
https://gist.github.com/garaemon/ed980a700edc935adb4b
This is the log

garaemon, 2015-01-26:
Could you run with DISPLAY=0:0?

garaemon, 2015-01-27:
I think this DISPLAY setting is reproducable