euslisp/jskeus #188: build failed jskeus (eusgl: not found)

furushchev, 2015-03-31:
failed to build jskeus after building euslisp successfly.

Maybe jskeus assumes that euslisp has `eusgl`, but euslisp does not.

link for logs when building jskeus and euslisp below:
https://gist.github.com/furushchev/0f58a84d2a8df645d18a

k-okada, 2015-03-31:
```

can't load font "*-courier-medium-r-*-8-*"can't load font
"*-courier-medium-r-*-10-*"can't load font
"*-courier-medium-r-*-12-*"can't load font
"*-courier-medium-r-*-14-*"can't load font
"*-courier-medium-r-*-18-*"can't load font
"*-courier-bold-r-*-12-*"can't load font
"*-courier-bold-r-*-14-*"can't load font
"*-courier-bold-r-*-18-*"can't load font
"*-courier-bold-r-*-24-*"can't load font
"*-times-medium-r-*-10-*"can't load font
"*-times-medium-r-*-12-*"can't load font "*-times-bold-r-*-12-*"can't
load font "*-times-bold-r-*-14-*"can't load font
"*-times-bold-r-*-18-*"can't load font "*-times-bold-r-*-24-*"can't
load font "lucidasans-bold-12"can't load font
"lucidasans-bold-14"can't load font
"*-Helvetica-Bold-R-Normal-*-12-*"can't load font
"*-Helvetica-Medium-R-Normal-*-12-*"X Error of failed request:
BadFont (invalid Font parameter)
  Major opcode of failed request:  56 (X_ChangeGC)
  Resource id in failed request:  0x0
  Serial number of failed request:  49
  Current serial number in output stream:  58
make[4]: *** [/home/leus/ros/hydro_parent/src/euslisp/Linux64/obj/glconst.o]  1
make[4]: : jobserver : -j1 ,  make  `+'
[100%] Built target compile_euslisp
[ [36meuslisp [0m]  [32m<== [0m '
[1m/home/leus/ros/hydro_parent/build/euslisp/build_env.sh
/usr/bin/make -j8 -l8 [0m' finished with return code ' [1m0 [0m' [0m
[ [36meuslisp [0m]  [1m [34m==> [0m '
[1m/home/leus/ros/hydro_parent/build/euslisp/build_env.sh
/usr/bin/make install [0m'  [30m [1min [0m '
[1m/home/leus/ros/hydro_parent/build/euslisp [0m' [0m
can't load font "*-courier-medium-r-*-8-*"can't load font
"*-courier-medium-r-*-10-*"can't load font
"*-courier-medium-r-*-12-*"can't load font
"*-courier-medium-r-*-14-*"can't load font
"*-courier-medium-r-*-18-*"can't load font
"*-courier-bold-r-*-12-*"can't load font
"*-courier-bold-r-*-14-*"can't load font
"*-courier-bold-r-*-18-*"can't load font
"*-courier-bold-r-*-24-*"can't load font
"*-times-medium-r-*-10-*"can't load font
"*-times-medium-r-*-12-*"can't load font "*-times-bold-r-*-12-*"can't
load font "*-times-bold-r-*-14-*"can't load font
"*-times-bold-r-*-18-*"can't load font "*-times-bold-r-*-24-*"can't
load font "lucidasans-bold-12"can't load font
"lucidasans-bold-14"can't load font
"*-Helvetica-Bold-R-Normal-*-12-*"can't load font
"*-Helvetica-Medium-R-Normal-*-12-*"X Error of failed request:
BadFont (invalid Font parameter)
  Major opcode of failed request:  56 (X_ChangeGC)
  Resource id in failed request:  0x0
  Serial number of failed request:  49
  Current serial number in output stream:  58
```

this is the problem, you have the reboot pc to enable font server (is
someone know better solution, please let me know ) or compile euslisp
with unset DISPLAY,

k-okada, 2015-04-21:
this has been solved by https://github.com/euslisp/EusLisp/pull/100

k-okada, 2015-04-21:
technically this is https://github.com/euslisp/EusLisp/issues/97

closed 2015-04-21 by k-okada