euslisp/jskeus #196: fullbody-ik で，あまり腰を動かさない方法

YuOhara, 2015-04-22:
@snozawa さんに先ほど聞いて，fullbody-ik1であまり胴の角度を動かさないように，
```
         (list (list (send *robot* :rleg :toe-p :child-link) 0)
-              (list (send *robot* :lleg :toe-p :child-link) 0))
+              (list (send *robot* :lleg :toe-p :child-link) 0)
+              (list (send *robot* :torso :waist-p :child-link) 0.1)
+              (list (send *robot* :torso :waist-y :child-link) 0.1)
+              )
```
のようにしてみました．
普通の目標位置へのIKだと腰が動きにくくなったのですが，腰を動かさないと解けないところでは，腰が大きく動いてしまいます．

IKが解けなくてもいいので，胴があまり動かないようにしたいです．一時的に関節のMAXを小さくするみたいな
ことができればいいと思うのですが，何か方法はありますでしょうか？

mmurooka, 2015-04-22:
:min や :max が、ルートリンクの6DOFの可動域です。

https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L392-L393
がデフォルトの値（で合ってる?）

https://github.com/jsk-ros-pkg/jsk_demos/blob/master/jsk_2015_06_hrp_drc/drc_task_common/euslisp/request-ik-from-marker.l#L549
のように使う

mmurooka, 2015-04-22:
直接話して上のは僕の勘違いで、
waist-pitchの可動域を制限したいということでした。

min-max がある軸は :max-joint-angle などを上書きできなくて、
どうすれば良いんでしたでしょうか。

snozawa, 2015-12-21:
moved to https://github.com/euslisp/jskeus/issues/314

closed 2015-12-21 by snozawa