euslisp/jskeus #205: want to hide reference axes in irtviewer

YuyaNagamatsu, 2015-05-06:
I want only to render euslisp robot models on irtviewer for illustrations in my papers or posters.

Is there any method to hide the reference axes in irtviewer, rather than translating the model and dragging the viewer to push out the axes ?

k-okada, 2015-05-06:
you can use draw-origin option, 
https://github.com/euslisp/jskeus/blob/master/irteus/irtviewer.l#L75

try 
``` 
(make-irtviewer :draw-origin nil :draw-floor t)
```

YuyaNagamatsu, 2015-05-06:
Thank you!
I got the axes hidden by (make-irtviewer :draw-origin nil) .

[image]
[image]

closed 2015-05-06 by YuyaNagamatsu

k-okada, 2015-05-08:
when we learn new thing, it is better update documentation, see https://github.com/euslisp/jskeus/pull/207/files for example and check 
http://euslisp-docs.readthedocs.org/en/latest/jskeus/jmanual/#_26 for document

YuyaNagamatsu, 2015-05-08:
Oh, I'm seeing the web manual for the first time. I did'nt know it.

It's already written about this option?
Can I edit any other documentation?

k-okada, 2015-05-09:
this option is not documented until
https://github.com/euslisp/jskeus/pull/207/files commit.

if you define function like
```
(defun hoge ()
  "this is document"
  (+ 1 2 3))
```
then that document string will appear on html or pdf, please find
"make-irtviewer" section on
http://euslisp.github.io/jskeus/jmanual-node174.html,
http://euslisp-docs.readthedocs.org/en/latest/jskeus/jmanual/#_26

Adding or editing new documentation is highly welcome.
https://github.com/euslisp/jskeus/commit/53cbce80963710ee1074ef8f9413999d7b58e13c
https://github.com/euslisp/jskeus/commit/a9bd7fa184621fa22afaaaf302002d7ab700bfd1

YuyaNagamatsu, 2015-05-09:
I see.
When I find lack of documentation, I will make pull request.