euslisp/jskeus #212: null space の使い方の例

iory, 2015-05-24:
はどこでしょうか。

k-okada, 2015-05-24:
https://github.com/euslisp/jskeus/pull/84
にもありますが，いかでしょうか．:step　を大きくしないと，目標姿勢は満たされているのですぐ変えてtきますが，大きくするとうごいているうちに目標姿勢が満たされないので，結構微妙な調整が必要です．

```
(load "irteus/demo/sample-robot-model.l")
(setq *robot* (instance sample-robot :init))
(send *robot* :reset-pose)
(objects (list *robot*))
(setq e (send *robot* :larm :end-coords :copy-worldcoords))
;; use :null-space                                                                                                                                                         
;; (:larm-shoulder-p :larm-shoulder-r :larm-shoulder-y :larm-elbow-p :larm-wrist-y :larm-wrist-r :larm-wrist-p) ;; (send-all (send *robot* :larm :joint-list)  :name)      
(send *robot* :larm :inverse-kinematics e :null-space (float-vector 0 0.3 0 0 0 0 0) :thre 2 :stop 200 :debug-view :no-message) ;;  larm-shoulde-r                         
;; use additional-nspace-list                                                                                                                                              
(print (send *robot* :larm :shoulder-r :joint-angle))
(send *robot* :larm :inverse-kinematics e :additional-nspace-list (list (list (send *robot* :larm :shoulder-r :child-link) -0.6)) :thre 2 :stop 200 :debug-view :no-messag\
e) ;;  larm-shoulde-r                                                                                                                                                      
(print (send *robot* :larm :shoulder-r :joint-angle))
(send *robot* :larm :inverse-kinematics e :null-space (float-vector 0 0.5 0 0 0 0 0) :thre 1 :stop 200 :revert-if-fail nil :debug-view :no-message) ;;  larm-shoulde-r     
(print (send *robot* :larm :shoulder-r :joint-angle))
```

iory, 2015-05-24:
ありがとうございます。
やってみます

closed 2015-05-24 by k-okada