euslisp/jskeus #216: weird thing happens when (copy-object (make-cube))

chiwunau, 2015-06-23:
when use (copy-object) to a cube, the color space seems to have some problem
let say
```
setq a (make-cube 100 100 100)
(send a :set-color :red)
(send *irtviewer* :objects a
(setq b (copy-object a))
(send b :translate #f(0 100 0))
(send *irtviewer* :objects b)
;;;both a and b show up here as red color
(send b :set-color :blue)
(send *irtviewer* :draw-object :flush t)
;;;b change to blue color, but a disappear
(send a :set-color :red)
(send *irtviewer* :draw-object :flush t)
;;;a show up again
```

k-okada, 2015-06-23:
I see, that because copy-object also copies gl context (https://www.opengl.org/wiki/OpenGL_Context)
https://github.com/euslisp/jskeus/blob/master/irteus/irtgl.l#L263, after you copied the object, you have to call :set-color to use new gl context for new object.

```
13.irteusgl$ describe a
plist=((:face-color . #f(0.996094 0.0 0.0))
       (:gl-displaylist-id (114706400 . 2))
       (:volume . 1.000000e+06)
       (:centroid . #f(0.0 -1.204171e-15 -1.455192e-15)))
rot=#2f((1.0 0.0 0.0) (0.0 1.0 0.0) (0.0 0.0 1.0))
pos=#f(0.0 0.0 0.0)
parent=nil
descendants=nil
worldcoords=#<coordinates #X686d108  0.0 0.0 0.0 / 0.0 0.0 0.0>
manager=#<body #X6658ee0 (:cube 100.0 100.0 100.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>
changed=nil
box=#<bounding-box #X685b118 #f(-50.8 -50.8 -50.8)/#f(50.8 50.8 50.8)>
faces=(#<face #X6928b90 :cube :bottom>
       #<face #X63b9118 :cube :top>
       #<face #X6928ad0 :cube :side 3 :y>
       #<face #X63b98b0 :cube :side 2 :x>
       #<face #X685cff0 :cube :side 1 :-y>
       #<face #X685cf78 :cube :side 0 :-x>)
edges=(#<edge #X685dcf8 #f(50.00 50.00 -50.00) #f(-50.00 50.00 -50.00)>
       #<edge #X6855d70 #f(50.00 50.00 50.00) #f(-50.00 50.00 50.00)>
       #<edge #X685d650 #f(50.00 -50.00 -50.00) #f(50.00 50.00 -50.00)>
       #<edge #X685d698 #f(50.00 50.00 -50.00) #f(50.00 50.00 50.00)>
       #<edge #X6855ea8 #f(50.00 -50.00 50.00) #f(50.00 50.00 50.00)>
       #<edge #X685d4a0 #f(-50.00 -50.00 -50.00) #f(50.00 -50.00 -50.00)>
       #<edge #X685d518 #f(50.00 -50.00 -50.00) #f(50.00 -50.00 50.00)>
       #<edge #X6855e30 #f(-50.00 -50.00 50.00) #f(50.00 -50.00 50.00)>
       #<edge #X685d0b0 #f(-50.00 50.00 50.00) #f(-50.00 50.00 -50.00)>
       #<edge #X685d128 #f(-50.00 50.00 -50.00) #f(-50.00 -50.00 -50.00)>
       #<edge #X685d170 #f(-50.00 -50.00 -50.00) #f(-50.00 -50.00 50.00)>
       #<edge #X6855de8 #f(-50.00 50.00 50.00) #f(-50.00 -50.00 50.00)>)
vertices=(#f(50.0 50.0 -50.0)
          #f(50.0 50.0 50.0)
          #f(50.0 -50.0 -50.0)
          #f(50.0 -50.0 50.0)
          #f(-50.0 50.0 -50.0)
          #f(-50.0 -50.0 -50.0)
          #f(-50.0 -50.0 50.0)
          #f(-50.0 50.0 50.0))
model-vertices=(#f(50.0 50.0 -50.0)
                #f(50.0 50.0 50.0)
                #f(50.0 -50.0 -50.0)
                #f(50.0 -50.0 50.0)
                #f(-50.0 50.0 -50.0)
                #f(-50.0 -50.0 -50.0)
                #f(-50.0 -50.0 50.0)
                #f(-50.0 50.0 50.0))
convexp=t
geometry::evertedp=nil
csg=((:cube 100.0 100.0 100.0))
nil
14.irteusgl$ describe b
plist=((:face-color . #f(0.0 0.0 0.996094))
       (:gl-displaylist-id (114706400 . 3))
       (:volume . 1.000000e+06)
       (:centroid . #f(0.0 -1.204171e-15 -1.455192e-15)))
rot=#2f((1.0 0.0 0.0) (0.0 1.0 0.0) (0.0 0.0 1.0))
pos=#f(0.0 100.0 0.0)
parent=nil
descendants=nil
worldcoords=#<coordinates #X69562f8  0.0 100.0 0.0 / 0.0 0.0 0.0>
manager=#<body #X68c8140 (:cube 100.0 100.0 100.0) 0.0 100.0 0.0 / 0.0 0.0 0.0>
changed=nil
box=#<bounding-box #X69077a8 #f(-50.8 49.2 -50.8)/#f(50.8 150.8 50.8)>
faces=(#<face #X694dbb0 :cube :bottom>
       #<face #X694d9b8 :cube :top>
       #<face #X694daf0 :cube :side 3 :y>
       #<face #X694da78 :cube :side 2 :x>
       #<face #X694dc70 :cube :side 1 :-y>
       #<face #X696b080 :cube :side 0 :-x>)
edges=(#<edge #X68c0ad0 #f(50.00 150.00 -50.00) #f(-50.00 150.00 -50.00)>
       #<edge #X69147e8 #f(50.00 150.00 50.00) #f(-50.00 150.00 50.00)>
       #<edge #X68c0d10 #f(50.00 50.00 -50.00) #f(50.00 150.00 -50.00)>
       #<edge #X6914770 #f(50.00 150.00 -50.00) #f(50.00 150.00 50.00)>
       #<edge #X69146f8 #f(50.00 50.00 50.00) #f(50.00 150.00 50.00)>
       #<edge #X68c0b90 #f(-50.00 50.00 -50.00) #f(50.00 50.00 -50.00)>
       #<edge #X68c0c50 #f(50.00 50.00 -50.00) #f(50.00 50.00 50.00)>
       #<edge #X68c0c08 #f(-50.00 50.00 50.00) #f(50.00 50.00 50.00)>
       #<edge #X6914830 #f(-50.00 150.00 50.00) #f(-50.00 150.00 -50.00)>
       #<edge #X68c0b18 #f(-50.00 150.00 -50.00) #f(-50.00 50.00 -50.00)>
       #<edge #X68c0cc8 #f(-50.00 50.00 -50.00) #f(-50.00 50.00 50.00)>
       #<edge #X69146b0 #f(-50.00 150.00 50.00) #f(-50.00 50.00 50.00)>)
vertices=(#f(50.0 150.0 -50.0)
          #f(50.0 150.0 50.0)
          #f(50.0 50.0 -50.0)
          #f(50.0 50.0 50.0)
          #f(-50.0 150.0 -50.0)
          #f(-50.0 50.0 -50.0)
          #f(-50.0 50.0 50.0)
          #f(-50.0 150.0 50.0))
model-vertices=(#f(50.0 50.0 -50.0)
                #f(50.0 50.0 50.0)
                #f(50.0 -50.0 -50.0)
                #f(50.0 -50.0 50.0)
                #f(-50.0 50.0 -50.0)
                #f(-50.0 -50.0 -50.0)
                #f(-50.0 -50.0 50.0)
                #f(-50.0 50.0 50.0))
convexp=t
geometry::evertedp=nil
csg=((:cube 100.0 100.0 100.0))
```

chiwunau, 2015-06-24:
Thank out very much, I will study it later.