euslisp/jskeus #217: fullbody-ikでもう少し中身を抜き出したサンプルが欲しい

yuki-asano, 2015-06-23:
野沢さんに相談した内容で、issueを切ります。

snozawa, 2015-06-23:
（もう少し具体的に。。。）

おそらく話を聞いているかんじでは
:inverse-kinematicsの中が比較的膨大で、数式にかいてあるヤコビアンの式まで
たどりにくい、ということのように思いました。

そうすると、jmanualのhttps://github.com/euslisp/jskeus/raw/master/doc/jmanual.pdfの
205ページくらいのサンプルコードが、`irteus/demo`以下にあって実行できるかんじになっていると良さそうでしょうか。

k-okada, 2015-06-23:
機情の学生なら，そのへんはソフトウェア第三でやっているはずで，
ヤコビアンの計算もでてきているから，一回資料見て勉強してもらうといいかもしれないですね．

yuki-asano, 2015-06-23:
そうですね。勉強が足りませんでした。
eusは色々用意されていて便利なのですが、中で何が起きているか分かりにくいので、
詳しく実装を理解したいときに、困ってしまったということでした。

k-okada, 2015-06-24:
https://github.com/jsk-ros-pkg/jsk_model_tools/issues/119#issuecomment-114385560
の人も同じように中がわからず困っているので，それが平均だとおもいますが．
観察していると，ソースを追っていくにどこで足切りしているかだけの問題にも
見えます．なので，もう一歩根気よく先まで見ていく，という以外に最終的な
解法はないはずですが，やはり，いきなり全部理解しようとするとよくわからず
困ってしまうはずで，簡単な例題から始めて理解を深めていくのがいいとおもいます．
６自由度のアームを用いたIK,ヤコビアンの計算の仕方，手先の速度の与え方，リンクリスト，
ターゲット，など基本的なところを抑えてみていけば，分かるんじゃないかと思います．
わかったところ，わからないところは，ソースの中にドキュメントとして入れてみて下さい．

yuki-asano, 2015-06-24:
今回は、重心ヤコビアンと足先ヤコビアンの併用の仕方が分かりませんでした。
足先の6DOFヤコビアンの計算や実装方法は梶田さんの本にも載っているので、
そこを見て自習することが出来たのですが、
梶田さん本に記載のない、
fullbody-ikの中身や、
重心ヤコビアンの求め方や基礎ヤコビアンとの併用方法
(欲しかったのは、足先位置と重心位置を入力としたときの足関節角度)については、
野沢さんに聞いてやり方及び参照するべき資料などを教えてもらいました。

まとめると、
今回分からなかったところは、以下です。
・重心ヤコビアンの実装方法
・重心ヤコビアンと基礎ヤコビアンの併用方法
・fullbody-ikの原理(中身は、足先6DOF + rootリンク(骨盤)の6DOFのik)

k-okada, 2015-06-24:
普通にikのコードを読んでいくと，union-link-listとか謎なfuncallが出てきます．
上のサンプルでは使わない部分でなんでこんなんあるんだろう，と思っているはずで，
その状態で，fullbody-ikを読んでいくと，ああこうやっているのか，なるほど，
うまく出来ているけど，やり過ぎでは？等など，と，なるはずです．
本当だとirtmodelにcog-xxみたいなコードがちらほら顔を出すので，脚なロボットで
ない人でも，ああ，このへんでなにかやっているな，というところまでは頭に入っている
はず，というのが，最大限に欲張った期待です．

XXが知りたい→その実装方法を調べる→IKのコードに行き着く
IKのコードの基礎を理解する←→その拡張方法を理解する←XXを知りたい

と理解するときの方向は２方向あるんですが，最初は個別のケースで理解していくしか
ないとはおもいますが，IKのコードの基礎，というか，（一般的な）ロボットのモデリング
の方法と，それをつかった応用方法がりかいできれば，ここで質問になっている重心ヤコビアン
も，https://github.com/jsk-ros-pkg/jsk_model_tools/issues/119 移動台車の例も，まぁ同じもの
として理解できるようになるとおもいます．

いずれにせよ，わかったことは
https://github.com/euslisp/jskeus/blob/master/irteus/irtdyna.l#L414
のコメントに追加して下さい．public扱いなものを
https://github.com/euslisp/jskeus/pull/175/files
を参考に(defun hoge () "hogeg" とドキュメントにしておけば，
http://euslisp-docs.readthedocs.org/en/latest/jskeus/jmanual/#_22
に説明がでてきます．

k-okada, 2015-06-24:
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L1791
ぐらいの説明があればいいのかな．