euslisp/jskeus #227: texture information is missing after converting eus model to collada

furushchev, 2015-07-09:
eusモデルに画像を貼ったものをirtcolladaを使ってcollada(DAE)に変換するとテクスチャ情報がなくなっています。

参考:
https://github.com/jsk-ros-pkg/jsk_demos/pull/988
の画像ではテレビの画面テクスチャと床のテクスチャが変換できていなくて黒くなっている

https://github.com/turtlebot/turtlebot/blob/indigo/turtlebot_description/meshes/sensors/kinect.dae#L17-L26
https://github.com/turtlebot/turtlebot/blob/indigo/turtlebot_description/meshes/sensors/kinect.dae#L36
https://github.com/turtlebot/turtlebot/blob/indigo/turtlebot_description/meshes/sensors/kinect.dae#L138-L142
https://github.com/turtlebot/turtlebot/blob/indigo/turtlebot_description/meshes/sensors/kinect.dae#L154

などを見るとテクスチャ画像ファイルをcolladaでロードするところまでは分かりそうですが、
eusで貼ったテクスチャのtexcoordをcolladaの形式にどう変換していいかわかりませんでした。
eusだとfacesetの `get :gl-texturecoords` にuvマップがあってそこから取れる気がしていますが、colladaではそのマップはどこに記述されるのでしょうか・・・

k-okada, 2015-07-09:
> jsk-ros-pkg/jsk_demos#988

https://github.com/turtlebot/turtlebot/blob/indigo/turtlebot_description/meshes/sensors/kinect.dae#L108
？？？

YoheiKakiuchi, 2015-07-09:
たぶん、あとここもですね。
https://github.com/turtlebot/turtlebot/blob/indigo/turtlebot_description/meshes/sensors/kinect.dae#L123-L124

メッシュだけなら、eus_assimpでいい感じに表示されるglverticesを作るとdaeへの変換は早いかもしれません。

furushchev, 2015-07-09:
@YoheiKakiuchi さんと相談した結果まとめ　（間違っていたら指摘してください）
- 今のeusモデル→urdfへの変換：
eus -> (irtcollada) -> dae -> (collada_urdf) -> urdf (meshとしてのdae)
となっているけど、テクスチャを貼る場合 irtcolladaでtexture情報が失われることを解決するだけでなくて、collada_urdfもtextureを無視しているようです。

furushchev, 2015-07-09:
irtcolladaについて、
- eusモデルからdaeのメッシュを作るときにテクスチャを貼るには, position, normalの他にtexcoordを指定する必要がある (cf. https://github.com/euslisp/jskeus/blob/master/irteus/irtcollada.l#L914-L959 )
- positionとnormalはfaceから`geo::body-to-triangles`で三角形になったあと、trianglesとして記述されるが、textureについては分割されない (cf. https://github.com/euslisp/jskeus/blob/master/irteus/irtcollada.l#L208 )

というところまではわかりました。この先どうしたらいいか悩んでいます。

snozawa, 2015-07-09:
そもそも誰か
http://wiki.ros.org/rtmros_common/Tutorials/ModelFileConversion
の場所にJSKで使ってる全モデル変換を一覧できるフロー図を作ってくれると助かります。

場所がrtmros_comonが問題そうであれば、このページはjsk_comonくらいに移してもよいかも。
rtmrosを使うひと、gazeboを使うひと、などこだわらずに、JSKの人が使ってるフォーマットと変換すべてを一望できて説明しているページが必須ですね。

k-okada, 2015-07-09:
三角分割しても頂点がふえるわけではないから，texture coordsをうまくあてはめればいいのでは？

garaemon, 2015-07-09:
vrmlのロボットだとこういう図があります
https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_calibration/hrp2jsknts_calibration/README.md#appendix

snozawa, 2015-07-09:
https://github.com/start-jsk/rtmros_tutorials/issues/84
のissueでは、urdfが今何種類も生成されてるけど、こんなに必要なのでしょうか、というのを説明したかったのですが、図があるとフローの中に.urdfファイルがたくさんあるというのが一目でわかりますね。