euslisp/jskeus #228: add example for preview controller and resolved momentum control

k-okada, 2015-07-10:
@snozawa which functions is "public" API that use should call in their program

preview control : https://github.com/euslisp/jskeus/blob/master/irteus/irtdyna.l#L1056 ??
momentum control: ??

k-okada, 2015-07-10:
cc: @terasawa

snozawa, 2015-07-13:
Example for Preview Controller is included in `demo/walk-motion.l`.
https://github.com/euslisp/jskeus/blob/master/irteus/demo/walk-motion.l#L195
irteus's preview contoller is oriented to generate COG trajectory and dynamics filtering.
I'll add documentation string to these classes later.
(@terasawa, adding documentation string is welcome!)

Currently, resolved momentum control is not implemented on irteus.
However, related calculation methods such as momentum jacobiand and angular momentum jacobian are already committed to irtdyna.l:
https://github.com/euslisp/jskeus/blob/master/irteus/demo/walk-motion.l#L195
Momentum jacobian is currently used as COG jacobiand and `:fullbody-inverse-kinematics` and so on.
These methods does not have documentation string currently.

snozawa, 2015-08-07:
@terasawa
I updated documentation and examples for preview control https://github.com/euslisp/jskeus/pull/254.
If you have any question and more request, please reopen this issue.

closed 2015-08-07 by snozawa