euslisp/jskeus #230: :draw-arrow bug?

snozawa, 2015-07-11:
I'd like to draw objects using `:draw-on` and draw arrow using `:draw-arrow`.

Some codes does not work.
For example, 

```
(objects (list (setq *bb* (make-cube 200 200 200))))

;; draw-on => draw-arrow : Work correctly
(progn  
(send *irtviewer* :draw-objects :flush nil)
(send-all (send *bb* :vertices) :draw-on :flush nil)
(send *irtviewer* :viewer :draw-arrow (float-vector 0 0 -100) (float-vector 0 0 100)) 
(send *irtviewer* :viewer :viewsurface :flush)
)

;; draw-arrow => draw-on : Does not work (vertices cannot be displayed)
(progn  
(send *irtviewer* :draw-objects :flush nil)
(send *irtviewer* :viewer :draw-arrow (float-vector 0 0 -100) (float-vector 0 0 100)) 
(send-all (send *bb* :vertices) :draw-on :flush nil)
(send *irtviewer* :viewer :viewsurface :flush)
)
```

k-okada, 2015-07-12:
try to fix this on https://github.com/euslisp/EusLisp/pull/119, but not
sure what kind of side effect will happen,
also we still have a good way to test these issue.

snozawa, 2015-07-13:
Sorry for late reply.
This PR (https://github.com/euslisp/EusLisp/pull/119) will fix the original problem:
https://github.com/euslisp/jskeus/issues/230#issue-94471185

However, sometimes surfaces of objects are inverted.
[image]

k-okada, 2015-07-13:
> Sorry for late reply.
> This PR (euslisp/EusLisp#119 <https://github.com/euslisp/EusLisp/pull/119>)
> will fix the original problem:
> #230 (comment)
>
> However, sometimes surfaces of objects are inverted.
> [image: cbox-inverted]
> <https://cloud.githubusercontent.com/assets/6102287/8642553/c575d16e-2960-11e5-89d0-57c509f785bd.jpg>

what do you mean by "sometmes" , run test program many times? or you have 
different test program?

snozawa, 2015-07-16:
Sorry for late reply...
My "sometimes" was incorrect word.

In failure case, always invalid displaying occurs and 
currently I could not find more small code to reproduce this.

Anyway, when I execute the original sample program, 
surface is inverted:
```
(objects (list (setq *bb* (make-cube 200 200 200))))

;; draw-on => draw-arrow : Work correctly
(progn  
(send *irtviewer* :draw-objects :flush nil)
(send-all (send *bb* :vertices) :draw-on :flush nil)
(send *irtviewer* :viewer :draw-arrow (float-vector 0 0 -100) (float-vector 0 0 100)) 
(send *irtviewer* :viewer :viewsurface :flush)
)

;; draw-arrow => draw-on : Does not work (vertices cannot be displayed)
(progn  
(send *irtviewer* :draw-objects :flush nil)
(send *irtviewer* :viewer :draw-arrow (float-vector 0 0 -100) (float-vector 0 0 100)) 
(send-all (send *bb* :vertices) :draw-on :flush nil)
(send *irtviewer* :viewer :viewsurface :flush)
)
```

Naoki-Hiraoka, 2018-09-29:
https://github.com/euslisp/jskeus/issues/32

表示の反転が起こるのは，
https://github.com/euslisp/EusLisp/pull/119
で削除している```(send self :3d-mode)```の中で，本当は重要な処理を行っていたためと考えられます．

```(send self :3d-mode)```中の
https://github.com/euslisp/EusLisp/blob/27f0d63baf58ecdaedeec891cd29f3bcb631604d/lisp/opengl/src/glview.l#L289-L290
で```(glEnable GL_DEPTH_TEST)```を行うのが重要で，これをしないと直方体の裏面が描画されてしまうことがあるようです．

https://github.com/euslisp/EusLisp/compare/master...Naoki-Hiraoka:push_pop_for_2d_draw
のように```(send self :3d-mode)```を復活させると，```:draw-on```と```:draw-arrow```の問題も発生せず，反転する問題も発生しなくなりました．

closed 2018-12-12 by k-okada