euslisp/jskeus #282: Support hand in irteus

snozawa, 2015-10-15:
I'd like to support hand in irteus.
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-utils.l#L117

I'll write example case later.

snozawa, 2015-11-21:
irtrobot.lのhttps://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L255にある:gripperメソッドは、
具体的にどこかのロボットで使ってるでしょうか？
pr2は別途pr2-utilsで定義してる気がするので（一部send-super）。

このgripperのメソッドをどうするかはありますが、
:handというメソッドで同様のものを作りたいのですが、いかがでしょうか。
また、ハンドの中身の扱いとしても
- ハンドがロボットの大本のモデルファイルに含まれてるもの：PR2, HRP2JSK
- ハンドが別ファイルで、Euslisp（やURDF）などでくっつけているもの：HRP2JSKNT, JAXON

で２とおりのサポートがいるような気がしています。
:handのメソッドとしては
```
(send *robot* :hand :rarm)
(send *robot* :hand :rarm :links)
(send *robot* :hand :rarm :joint-list)
(send *robot* :hand :rarm :angle-vector)
(send *robot* :hand :rarm xx) ;; その他ハンド自体がインスタンスの場合（HRp2JSKNT, JAXON）その固有のメソッド。reset-poseなど
```
で、:rarmの箇所は`:rarm`, `:larm`, `:arms`くらいをつかいたいです。

メソッド追加の方法としては、:handというメソッドを使って、互換性たもつために:gripperメソッドでwrapしておくなどを考えてます

k-okada, 2015-11-21:
https://github.com/jsk-enshu/robot-programming/blob/964d3c31f067960cbc4fcd532ce93eb603c8dcc9/dynamixel_7dof_arm/euslisp/dxl-7dof-arm-interface-common.l#L154

とかで使っている？

(defmethod (:hand (&rest args) (forward-message-to self :gripper args)))

みたいなのではダメのかな．

snozawa, 2015-11-22:
> みたいなのではダメのかな．

はい、
- :gripperを定義して、:handは:gripperのforward-message-toにするか
- :handを定義して、:gripperは:handのforwared-message-toにするか

だと思いましたので、それでも良いです。
あとは、今の`:gripper`の中にハンドモデルが与えられた場合にそれを使うようにすればよさそうです。

YoheiKakiuchi, 2015-11-22:
forward-message-toするとして、新しく:handメソッドが必要なのかな？

snozawa, 2015-11-23:
これはどちらかというとメソッドのnamingの問題で，
JAXONやHRP2JSKNTなどグリッパというよりはハンドというほうが
適していそうなものを手首から先に搭載してるロボットで，
ハンドにアクセスするメソッド名は`:gripper`より`:hand`が良いと思いました．
もちろん，命名則はあるにせよ同じようなものが２こできるのはよくないので，
（歴史的に先にirteusにできた）`:gripper`に統一しておくべき，ということでしたらそうします．

snozawa, 2015-12-25:
こちらですが，やはり`:hand`でなくて`:gripper`の方がよいでしょうか．

k-okada, 2015-12-25:
jsk-enshuも直してくれればどちらでもいいですが，今見ると，
```
(send *robot* :hand :rarm :links)
(send *robot* :rarm :links)
(send *robot* :head :links)
```
とみると，:lhand, :rhand であるべき，になっちゃいますね．

snozawa, 2015-12-25:
> とみると，:lhand, :rhand であるべき，になっちゃいますね．

確かに、そうですね。
`(send *robot* :rhand)`も`(send *robot* :hand :rarm)`もどちらもありえそうです。

さらによくみると
https://github.com/euslisp/jskeus/commit/239047b1099dba5b9f24a5cb2b79630f43f02c8c
で
```
(send *robot* :rarm :gripper)
```
も可能になってますね。
これは今使われてますでしょうか。

k-okada, 2016-01-29:
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/bd2e145626f72f40512d682e8d4f52ff97bd3f76/pr2eus/pr2-interface.l#L118
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/bd2e145626f72f40512d682e8d4f52ff97bd3f76/pr2eus/pr2-interface.l#L232
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/bd2e145626f72f40512d682e8d4f52ff97bd3f76/pr2eus/pr2-interface.l#L274
とありそうです．