euslisp/jskeus #283: irtviewerをoffscreenで描画/取得したい

furushchev, 2015-10-19:
irtviewerをxwindowを使わずに描画して画像を取得したいです。
どのように変更を加えていくのがいいでしょうか？

```lisp
(pr2)
(setq *ctx* (make-irtviewer :no-x))
(send *ctx* :objects (list *pr2*))
(setq img (send (send *ctx* :draw-objects) :write-to-image))
(write-image-file img "/path/to/image")
```

YoheiKakiuchi, 2015-10-19:
これはどういう用途で必要になるかな？

furushchev, 2015-10-19:
画面のないPCでirtviewerの画面を取得したい時にほしいです。
その画像をdbなどに保存する用途で僕は使いたいです。

YoheiKakiuchi, 2015-10-19:
offscreen でもxwindowがあるかどうかで対応が違うのだろうと思います。
X無しでopenglを使う方法は分からないです。

- Xwindowは上がっていて、windowを出さないで描画する
- Displayが無い、(通常ではXが上がっていない)

なので、ディスプレイがないPCでディスプレイを必要としないxserverを立ち上げて、
画像が取得できるか見てみるのだと思います。
https://en.wikipedia.org/wiki/Xvfb

k-okada, 2015-10-19:
opengl can support offscreen rendering
http://stackoverflow.com/questions/12157646/how-to-render-offscreen-on-opengl

garaemon, 2015-10-22:
virtualgl http://www.virtualgl.org/ https://github.com/VirtualGL/virtualgl