euslisp/jskeus #284: joint の座標系の扱いについて

s-noda, 2015-10-19:
既視感があるので既出だったら申し訳ないのですが、
joint の座標系の扱いについてのポリシーみたいなものについてお聞きしたいのです。

irtmodel.l, irtdyna.l を見ると joint の座標系の表現が二種類に揺れている気がします。

```
(setq a (send (send (send joint :parent-link) :copy-worldcoords) :transform (joint . default-coords))
```
```
(setq b (send (send joint :child-link) :copy-worldcoords))
```

例としては、

https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L124
を見ると、世界座標系での回転軸の計算として ``joint . axis``` を ```a```で変換したものが使われていて、

https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L135
を見ると、ヤコビアン計算時の joint の位置として ```b``` の ```:worldcoords``` が使われています。

手元の hrp2 だと ```a``` と ```b``` は ```:init-pose``` のときに一致しているようなので、```b```を使っても許される状況があるのだと思いますが、ここは ```a``` に統一すべきではないでしょうか？

snozawa, 2015-10-19:
> 手元の hrp2 だと a と b は :init-pose のときに一致しているようなので、bを使っても許される状況があるのだと思いますが、ここは a に統一すべきではないでしょうか？

jointの座標系はeusではないことになっています。
基本的にはforward kinematicsが定義されてるcascaded-coordsの子クラスであるbodyset-linkが
リンクとしてリンク座標系をもちます。
概ね@s-nodaさんのいうjoint座標系は
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L1140
のように"parent-link座標系にdefault-coordsを掛け合わせたものがjoint-angle=0のときのchild-link座標系"としたものをさしてるものに相当するんだけど、これはあってるかな？

質問としては
- `a`パターンになってないと問題がある箇所があるかな？
- https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L135の箇所以外に何か修正が必要な箇所があるかな？

snozawa, 2015-10-19:
あと質問一個追加で、
- そもそもどういうものを指してjoint座標系といってますか？これによっては、どちらに統一すべきがかわるかもしれないし、かわらないかもしれない。

s-noda, 2015-10-19:
回答ありがとうございます．補足します．

> そもそもどういうものを指してjoint座標系といってますか？これによっては、どちらに統一すべきがかわるかもしれないし、かわらないかもしれない。

回転軸の座標系です．
回転軸のある位置と，回転軸の姿勢を表す coordinates．

> parent-link座標系にdefault-coordsを掛け合わせたものがjoint-angle=0のときのchild-link座標系

はい．これの ``` parent-link座標系にdefault-coordsを掛け合わせたもの``` に統一したほうがいいのではないかという issue です．
というか，これと ```joint-angle=0のときのchild-link座標系``` が常に一致することを期待する実装は自由度が足りないのではないかということがお尋ねしたいことです．

なぜなら，回転軸と child-link の座標系を別々にしたいような場合，
具体的に僕が昨日やってみたことですが，
既存のロボットモデルを改造して，ルートリンクを左足のEFにしたいような場合に，左足のリンク系の親子関係はすべてひっくり返るので， child-link の回転軸 → 元のparent-link の回転軸となって， ```parent-link座標系にdefault-coordsを掛け合わせたものがjoint-angle=0のときのchild-link座標系``` にできなくなるきがします．

> aパターンになってないと問題がある箇所があるかな？

ないと思います．

> https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L135の箇所以外に何か修正が必要な箇所があるかな？

僕が気づいたのは

```
(defun calc-jacobian-rotational
```
```
(defun calc-inertia-matrix-rotational
```
```
(defmethod rotational-joint
  (:calc-spacial-velocity-jacobian
```

ですが，
こうすると慣性行列とかの計算も変わってすごく面倒な問題になる気がしてきました．．．

snozawa, 2015-10-20:
> というか，これと joint-angle=0のときのchild-link座標系 が常に一致することを期待する実装は自由度が足りないのではないかということがお尋ねしたいことです．

これは難しいんだけど、
rotational-joint, linear-joint, 6dof-joint, sphere-jointなど全ての関節で
joint-angle=0のときchild-link＝parent-linkt*default-coordsになります。
むしろdefault-coordsの定義がそういうものになってます。

> 常に一致することを期待する実装は自由度が足りないのではないかということがお尋ねしたいことです．

これは一致しないようにしたいという具体的な要望がありますか？

ちなみに、
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L135
を@s-nodaさんがいう`a`方式に統一しても問題ないと思います。
その理由は、ここはrotational-jointを家庭している部分で、かつ
rotational-jointではかならずparent-link+default-coordsのworldposとchild-linkのworldposが
一致するので、置き換えてもよさそう、ということです
（parent-link+default-coordsとcil-dlinkの違いは、1軸回転関節の角度分しかないため）

s-noda, 2015-10-20:
はい。あとでPRします。

```
(defun calc-jacobian-rotational
```
```
(defun calc-inertia-matrix-rotational
```
```
(defmethod rotational-joint
  (:calc-spacial-velocity-jacobian
```

も同様なので、 a に統一して問題ないと思います。
rotational-joint で、 child-link の :worldpos を使っているという意味で

> これは一致しないようにしたいという具体的な要望がありますか？

左足のEFをルートリンクにしたロボットモデルを作った時に、回転軸とリンクの座標系が違ったものになるような rotational-joint を定義したくなりました。

snozawa, 2015-10-20:
> 既存のロボットモデルを改造して，ルートリンクを左足のEFにしたいような場合に，左足のリンク系の親子関係はすべてひっくり返るので， child-link の回転軸 → 元のparent-link の回転軸となって， parent-link座標系にdefault-coordsを掛け合わせたものがjoint-angle=0のときのchild-link座標系 にできなくなるきがします．

ルートリンクを左足EFにするようなものは
```(load "irteus/demo/demo.l")```
```(full-body-ik :use-leg t)```
みたいなことかな？

リンクリストが逆順になる場合にヤコビアンを計算するものが
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L1130
にあります。
他のリンク系列をたどる系の計算は、逆順計算に対応してるかしてないか確認がいるかも
（多分あまりできてない）

s-noda, 2015-10-20:
ああ、前々からコレが何なのか疑問でしたが、そういう意図だったのですね。
もう大体動くものをつくってしまいました。。。

ところで別件ですが、あってるか自信ないですが、この child-reverse を計算する箇所、計算量的に問題がありませんか？
数十自由度だと問題ないのかもしれませんが、
jacobian の計算量はリンク長に対して O(n) ですが、
n 回のループの中の処理が O(1) で無い気がして、特に child-reverse まわりで
position や link-list の全要素に対する再帰計算 (find-parent) が中でさらに link-list にたいして find している?