euslisp/jskeus #286: calc-walk-pattern-from-footstep-listの目標重心位置が発散することがある

eisoku9618, 2015-10-21:
``go-pos-params->footstep-list``で生成したfoot step listを元に``calc-walk-pattern-from-footstep-list``で軌道生成しているのですが，``calc-walk-pattern-from-footstep-list``の目標重心軌道の最初の方の値に0ではなく大きな値が入ることがあって，最初の方の歩行軌道が変になることがあります．

以下の関数で再現できます．

```lisp
(test :interval 10 :custom-initial-pose t :check-centroid nil) ;; 目標重心軌道の最初の方に大きめの値が入る現象が再現する
(test :interval 10 :custom-initial-pose nil :check-centroid nil) ;; 目標重心軌道の最初の方にやや大きめの値が入る現象が再現する
(test :interval 10 :custom-initial-pose t :check-centroid t) ;; 再現しない気がする
```

```lisp
(defun test (&key (custom-initial-pose t) (fname-prefix "/tmp/a") (interval 1) (check-centroid nil))
  ;; generate *robot*                                                                                                                                                                                                                                                                                                                                                                                                                     
  (unless (boundp '*robot*)
    (load "package://hrpsys_ros_bridge_tutorials/euslisp/samplerobot-interface.l")
    (setq *robot* (samplerobot))
    (objects (list *robot*)))
  ;; initial-pose                                                                                                                                                                                                                                                                                                                                                                                                                         
  (progn
    (if custom-initial-pose
        (send *robot* :angle-vector #f(0.324866 -23.4957 -0.046053 47.5977 -24.3293 -0.558108 29.9965 -0.001226 -0.000706 -99.9755 9.00132 -6.49726 36.4998 -0.319302 -23.7343 0.063721 47.5631 -23.9359 0.522499 29.9965 0.001213 0.000702 -99.9755 -9.00132 -6.49726 -36.4998 -0.007121 0.000209 -3.915073e-06))
      (progn
        (send *robot* :reset-manip-pose)
        (send *robot* :fix-leg-to-coords (make-coords))
        (send *robot* :move-centroid-on-foot :both '(:rleg :lleg))))
    (send *robot* :fix-leg-to-coords (make-coords))
    (send *irtviewer* :draw-objects)
    (if check-centroid (format t "centroid : ~A~%" (send *robot* :centroid)))
    )
  (let (rs-list (idx 0))
    (when (y-or-n-p "start?")
      ;; calc-walk-pattern-from-footstep-list                                                                                                                                                                                                                                                                                                                                                                                             
      (setq rs-list
            (send *robot* :calc-walk-pattern-from-footstep-list
                  (send *robot* :go-pos-params->footstep-list 0 100 0)
                  :debug-view :no-message
                  :init-pose-function #'(lambda () ())
                  :solve-angle-vector-args (list
                                            :min (float-vector -1e5 -1e5 -1e5 -180 -180 -180)
                                            :max (float-vector +1e5 +1e5 +1e5 +180 +180 +180)
                                            :absolute-p t
                                            :debug-view :no-message
                                            :cog-null-space nil
                                            :additional-weight-list (list (list (send *robot* :torso :waist-y :child-link) 0.0))
                                            :root-link-virtual-joint-weight (float-vector 0.5 0.5 0.5 0.0 0.0 0.0)
                                            :cog-gain 5.0 :centroid-thre 50
                                            :init nil
                                            )
                  ))
      ;; save images                                                                                                                                                                                                                                                                                                                                                                                                                      
      (mapcar #'(lambda (rs)
                  (send *robot* :angle-vector (cadr (memq :angle-vector rs)))
                  (send *robot* :move-coords (cadr (memq :root-coords rs)) (car (send *robot* :links)))
                  (send *irtviewer* :draw-objects)
                  (if (= (mod idx interval) 0)
                      (send *irtviewer* :viewer :viewsurface :write-to-image-file (format nil "~A-~0,3d.png" fname-prefix idx)))
                  (incf idx))
              rs-list)
      )
    )
  )
```

再現した時の画像が以下になります．
y方向に100mm移動する指令を送たところ，最初の方の目標重心軌道のy成分が大きくて，最初の段階でy方向に重心が持って行かれてしまっています．

[image]

eisoku9618, 2015-10-21:
``calc-walk-pattern-from-footstep-list``の返り値を実機に送るときに，最初の値がおかしいと実機も変な姿勢になってしまうのですが，``calc-walk-pattern-from-footstep-list``の使い方で何か足りていないところがありますでしょうか？

snozawa, 2015-10-21:
こちらの手元でも再現できました。
見てみます。

snozawa, 2015-12-26:
別件で調べていたら直しかたがわかったので
https://github.com/euslisp/jskeus/pull/327
で直しました。

closed 2015-12-26 by snozawa

eisoku9618, 2015-12-26:
ありがとうございます．