euslisp/jskeus #287: pr2-robotに付いているcamera-modelでdraw-objectsしたい

furushchev, 2015-10-26:
pr2-robotのインスタンスに付いているカメラの画角で写っている自分自身の姿勢を見たいのですが、やり方がよくわからず、
https://github.com/euslisp/jskeus/blob/master/irteus/demo/sample-camera-model.l#L28
を参考にして、

```lisp
(pr2)
(send *pr2* :rarm :inverse-kinematics (make-coords :pos #f(700 0 1200) :rpy (list 0 -pi/2 0)))
(setq cam (send *pr2* :wide_stereo-left))
(send cam :draw-objects (list *pr2*))
(setq img (send cam :get-image))
```

として見ましたところ、

```lisp
4.irteusgl$ (send cam :draw-objects (list *pr2*))
/home/furushchev/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :viewsurface in (send cam :draw-objects (list *pr2*))
```

とエラーが出るようです。
どのようにしたらいいでしょうか？

YoheiKakiuchi, 2015-10-26:
https://github.com/euslisp/jskeus/blob/master/irteus/demo/sample-camera-model.l
これを参考に試してみて下さい。
camera-modelにあとからcreate-viewerできるようにした方がいいように思えてきました。

furushchev, 2015-10-26:
なるほど、viewerを追加する方法を教えていただけないでしょうか？

なんとなく

```lisp
(setf (camera-model-img-viewer cam) (view :width (send cam :width) :height (send cam :height)))
(send cam :draw-objects (list *pr2*))
```

とするとそれらしい画像が得られましたが、これで合っているでしょうか？

[image]

furushchev, 2015-10-26:
もうひとつ質問です。

```lisp
irteusgl$ send cam :screen-point (send (send *pr2* :rarm :end-coords) :worldpos)
#f(263.312 292.362)
```

とすると得られるuvは、armのend-coordsが写っているカメラの画像中の座標で合っているでしょうか？

furushchev, 2015-10-26:
それと、現在ロボットは

[image]

というアームの位置なのですが、上のカメラ画像は左右反転している気がします。。。

YoheiKakiuchi, 2015-10-26:
> なるほど、viewerを追加する方法を教えていただけないでしょうか？

それは、ちょっとコード書く必要あると思います。

まずは、make-camera-from-param で正しい画像が出せるか見てみるのがいいかと思います。

furushchev, 2015-10-26:
@YoheiKakiuchi

> まずは、make-camera-from-param で正しい画像が出せるか見てみるのがいいかと思います。

 勉強不足でカメラパラメータにあまり詳しくないので、rosのcamera_infoからimportしたcamera_modelで検証したいのですが、
`make-camera-from-ros-camera-info`を代わりに使っても検証可能ですか？

YoheiKakiuchi, 2015-10-26:
`make-camera-from-ros-camera-info` の中身は、make-camera-from-paramを使っているので、
これに、:create-viewer t を追加してやってみて下さい。

furushchev, 2015-10-26:
[image]

```lisp
12.irteusgl$ setq a (arrow)
13.irteusgl$ send a :translate (send (send *pr2* :rarm :end-coords) :worldpos)
#<arrow-object #X9a74690 arrow  699.87 -0.031 1199.959 / 0.0 0.0 0.0>
14.irteusgl$ objects (list *pr2* a)
15.irteusgl$ send-all (send *pr2* :cameras) :draw-objects (list *pr2* a)
```

やはり左右が反転しているようです。

あと、

> とすると得られるuvは、armのend-coordsが写っているカメラの画像中の座標で合っているでしょうか？

この仮説は正しいですか？

furushchev, 2015-10-26:
あれ、してない？

YoheiKakiuchi, 2015-10-26:
> 得られるuvは、armのend-coordsが写っているカメラの画像中の座標で合っているでしょうか？

これは、そうなっているはずです。

furushchev, 2015-10-26:
なるほど、とすると、pr1012のwide_stereo/leftを例に実ロボットのカメラ画像にeusのuvをプロットすると、

[image]

のように実機の手先位置とずれてしまうようです。
実機のbagデータからpr2.lを作りなおしてやっても、pr1040で同じ結果になったのですが、
これはbagデータからpr2.lを作っている部分が間違っているという事でしょうか？

YoheiKakiuchi, 2015-10-26:
座標系のとりかたの問題かもしれないので、手計算で検算してみるのがいいかと思います。

furushchev, 2015-10-26:
```lisp
8.irteusgl$ (send *tfl* :lookup-transform "wide_stereo_optical_frame" "base_footprint" (ros::time 0))
#<cascaded-coords #Xb7bcf30 wide_stereo_optical_frame  18.627 1324.362 -132.392 / 1.379 -1.519 0.2>
9.irteusgl$ (send (send *pr2* :wide_stereo-left) :transformation *pr2*)
#<coordinates #X97f7bc0  21.262 1330.071 -75.543 / 2.152 -1.556 -0.573>
```

これは一致しているべきですか？

furushchev, 2015-10-26:
@YoheiKakiuchi さんに見てもらった結果実機の画像はrectify前を使っていることがわかりました。
引き続きデバッグしてくださるようです。

furushchev, 2015-10-29:
これについては出来たのでcloseします。

closed 2015-10-29 by furushchev

k-okada, 2015-10-29:
マニュアルかサンプルプログラムにやり方を書いておきたい所

k-okada, 2015-11-01:
確認ですが，
https://github.com/euslisp/jskeus/issues/287#issuecomment-151028349
で，「u, v」をプロットする，というのは `:screen-pos` をつかっているんだよね？これは
- 結果実機の画像はrectify前を使っていることがわかりました。
- 引き続きデバッグしてくださるようです。
rectify 済みの画像を使うだけで，上の問題はなおったでしょうか？ #289 はシミュレション画像を作っている部分で，`screen-pos`とは関係ない気がしますが．．．

furushchev, 2015-11-01:
> rectify 済みの画像を使うだけで，上の問題はなおったでしょうか？

はい、これで直っているように見えます。問題はrectify済画像を使うということで、デバッグの過程でシミュレーション画像がずれていることも見つけたと言うだけだと思っています。
（issueも2つに分けました。）

uv変換は

```lisp
(send *pr2* reset-coords)
(setq cam (*pr2* :wide_stereo-left))
(setq uv (send cam :screen-point (make-coords :pos #f(700 0 1200))))
(setq msg (instance geometry_msgs::PointStamped :init))
(send msg :header :frame_id "wide_stereo_optical_frame")  ;; プロットしたい画像のframe_id
(send msg :point :x (elt uv 0))
(send msg :point :y (elt uv 1))
(ros::publish "target" msg)
```

プロットは既にあるc++プログラムで行っていますが、
https://github.com/jsk-ros-pkg/jsk_demos/blob/master/elevator_move_base_pr2/src/color_point_detector.cpp#L116
単純に受け取ったpointを`cv::circle`で書いているだけです。

k-okada, 2015-11-01:
了解です

k-okada, 2015-11-12:
closed from #290