euslisp/jskeus #295: get-image でmismatch argument になる

k-okada, 2015-11-01:
```
$ (load "irteus/demo/sample-robot-model.l")                                                                                  
$ (setq *robot* (instance sample-robot :init))                                                                                           
$ (setq *cam* (send *robot* :camera "left-camera"))
$ (send *cam* :get-image)
irteusgl 0 error: mismatch argument in (send *cam* :get-image)
```
なんだけど，
https://github.com/euslisp/jskeus/issues/287#issue-113282228
はエラーは報告されていない．なぜ？どのバージョン使っている? @furushchev

k-okada, 2015-11-01:
https://github.com/euslisp/jskeus/pull/268 が動いていたのか？

furushchev, 2015-11-01:
僕の方では

```
$ roseus                    ⮂  
configuring by "/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql 
;; extending gcstack 0x6497390[16374] --> 0x6799460[32748] top=395f
irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb irtgl irtviewer 
EusLisp 9.12(27da47d 1.0.9) for Linux64 created on musca(Sun Nov 1 18:26:34 JST 2015)
roseus ;; loading roseus("9712e53") on euslisp((9.12 musca Sun Nov 1 18:26:34 JST 2015 27da47d 1.0.9))
eustf roseus_c_util 1.irteusgl$ load "irteus/demo/sample-robot-model.l")
t
2.irteusgl$ setq *robot* (instance sample-robot :init))      
#<sample-robot #X6734e00 sample-robot  0.0 0.0 0.0 / 0.0 0.0 0.0>
3.irteusgl$ setq *cam* (send *robot* :camera "left-camera"))
#<camera-model #X6cfa5e8 left-camera (:cylinder 10.0 20.0 12) 0.0 30.0 305.0 / 0.0 1.571 0.0>
4.irteusgl$ send *cam* :get-image)
/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :viewsurface in (send *cam* :get-image)
```
という別のエラーがでます。

```
$ rosversion jskeus         ⮂  
1.0.9
```

furushchev, 2015-11-01:
```lisp
roseus                    ⮂  
configuring by "/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql 
;; extending gcstack 0x49ed390[16374] --> 0x4cef460[32748] top=395f
irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb irtgl irtviewer 
EusLisp 9.12(27da47d 1.0.9) for Linux64 created on musca(Sun Nov 1 18:26:34 JST 2015)
roseus ;; loading roseus("9712e53") on euslisp((9.12 musca Sun Nov 1 18:26:34 JST 2015 27da47d 1.0.9))
eustf roseus_c_util 1.irteusgl$ load "irteus/demo/sample-robot-model.l")  
t
2.irteusgl$ setq *robot* (instance sample-robot :init))     
#<sample-robot #X4c8ae00 sample-robot  0.0 0.0 0.0 / 0.0 0.0 0.0>
3.irteusgl$ setq *cam* (send *robot* :camera "left-camera"))
#<camera-model #X52505e8 left-camera (:cylinder 10.0 20.0 12) 0.0 30.0 305.0 / 0.0 1.571 0.0>
5.irteusgl$ send *cam* :create-viewer
#<viewer #X5090058 left-camera_view>
6.irteusgl$ send *cam* :get-image
#<color-image24 #X4e64c30 320x240>
```

`get-image`の前に`create-viewer`するとうまく行きます。

k-okada, 2015-11-01:
https://github.com/euslisp/jskeus/pull/268/files
でAPI変わっている．これをつかっていると1.0.10以降に思えるけど．．．

furushchev, 2015-11-01:
jskeusはsourceから入れていて、上のコマンド実行直前に
`git pull origin master && catkin bt`しました。
package.xmlには1.0.9と書かれています。
git logをすると、#268の変更はもう適用されているように見えます。

k-okada, 2015-11-01:
なるほど．
1) https://github.com/euslisp/jskeus/issues/48 の使い方は出来ない．:get-image vrwではなく:get-image
2) :get-image する前に :draw-objects する　（https://github.com/euslisp/jskeus/blob/master/irteus/demo/sample-camera-model.l#L28　）

コレ以外に，
https://github.com/k-okada/jskeus/commit/5c7317b9e219f6fe2344b554c0bbb2c6ff8d5c8d
で
send cam :create-viewer vwr
として既存のviwerをセットできるようにしましたが，上の1)の変更の副作用とも言える

あと，package.xml は euslisp/jskeus由来ではなくてjsk.rosinstall 的なものでjskeus-releaseから持ってきているのでふるいのはしょうがない．
`EusLisp 9.12(27da47d 1.0.9) for Linux64 created on musca(Sun Nov 1 18:26:34 JST 2015)`
を見るといいはずだが，ここもふるいなぁ．

closed 2015-11-01 by k-okada