euslisp/jskeus #303: :fullbody-inverse-kinematicsが解くたびに遅くなる

garaemon, 2015-11-16:
GC?

テストコード
```
(bench(dotimes (i 10)
 (let* ((mt (list (send *robot* :larm :end-coords)
                  (send *robot* :lleg :end-coords)
                  (send *robot* :rleg :end-coords)))
        (ll (mapcar #'(lambda (move-target)
                        (send *robot* :link-list (send move-target :parent)))
                    mt)))
   (send *robot* :reset-manip-pose)
   (send *robot* :fullbody-inverse-kinematics
         (list (send *robot* :larm :end-coords :copy-worldcoords)
               (send *robot* :lleg :end-coords :copy-worldcoords)
               (send *robot* :rleg :end-coords :copy-worldcoords))
         :link-list ll
         :move-target mt
         :target-centroid-pos (apply #'midpoint 0.5 (send  *robot* :legs :end-coords :worldpos))
         :warnp nil
         :stop 50 :min-loop 49))))
```

実行結果
```
;; time -> 1.72598[s]
;; time -> 1.99427[s]
;; time -> 1.97822[s]
;; time -> 2.06796[s]
;; time -> 2.11839[s]
;; time -> 2.21805[s]
;; time -> 2.2754[s]
;; time -> 2.30082[s]
;; time -> 2.36751[s]
;; time -> 2.41442[s]
;; time -> 2.60414[s]
```

garaemon, 2015-11-16:
続けていると、8secくらいまでおそくなります。

`;; extending gcstack 0x6c43d10[65466] --> 0xb7567f0[130932] top=ffb9`とでるので、GCかもしれない

garaemon, 2015-11-16:
もしかして`:assoc`でvirtual-jointがどんどん増えている?

garaemon, 2015-11-17:
`*robot*`を作りなおすと早くなる(リセットされる)

YoheiKakiuchi, 2015-11-17:
```(send *robot* :get :ik-draw-on-params)```　が増え続けていますね。
debug-viewにするとリセットされるようだけど、それ以外では増え続けているように見える。
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L2131

garaemon, 2015-11-17:
あたりですね

snozawa, 2015-12-21:
Solved via https://github.com/euslisp/jskeus/pull/323.

closed 2015-12-21 by snozawa