euslisp/jskeus #305: jmanualにjointクラスのインスタンスを引数に取るときの書き方が書かれていない

yurikadoi, 2015-11-18:
http://euslisp-docs.readthedocs.org/en/latest/jskeus/jmanual/#_22

rotational-jointは :init の説明が有り:axisを含め３つのキーワド引数の説明がありますが、jointは:initがありません。
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L47
にかかれていますが、どういう引数をとればいいか？:parent-link, :child-link が書いていません

cc: @k-okada

wkentaro, 2015-11-21:
euslispでループ内でcontinueする方法に関するPull Request
https://github.com/euslisp/EusLisp/pull/152#issuecomment-158594592
のようにPRを送るとjmanualを変更できます。

上はeuslispですが、jskeusだとここだと思います。
https://github.com/euslisp/jskeus/tree/master/doc

yurikadoi, 2015-11-22:
了解しました。ありがとうございます。

furushchev, 2016-07-13:
@yurikadoi If this issue is already solved, could you please close this issue?
And we will greatly appreciate if you contribute to add to manual about this issue and send your contribution as pull request to us.