euslisp/jskeus #306: wheel-joint のサンプルはありますでしょうか？

yurikadoi, 2015-11-18:
```
mech-user@test1-pc:~/2015winter/2015-soft3/20151104/src/enshu_20151104$ cd $EUSDIR
mech-user@test1-pc:/opt/ros/indigo/share/euslisp/jskeus/eus$ ls
Linux64  contrib  doc  include  irteus  lib  lisp  models
mech-user@test1-pc:/opt/ros/indigo/share/euslisp/jskeus/eus$ cd irteus/demo/
mech-user@test1-pc:/opt/ros/indigo/share/euslisp/jskeus/eus/irteus/demo$ ls
closed-loop.l    full-body-ik.l   particle.l                   special-joints.l
crank-motion.l   hand-grasp-ik.l  sample-arm-model.l           walk-motion.l
demo.l           hanoi-arm.l      sample-hand-model.l
dual-arm-ik.l    hanoi.l          sample-multidof-arm-model.l
dual-manip-ik.l  null-space-ik.l  sample-robot-model.l
mech-user@test1-pc:/opt/ros/indigo/share/euslisp/jskeus/eus/irteus/demo$ grep wheel *
```

あと、`cd $EUSDIR`を初めて知りました。どこにかいてあったでしょうか。

wkentaro, 2015-11-21:
EUSDIRはここ
https://github.com/euslisp/jskeus/blob/master/README.md#362-downloading-and-building-euslisp
でbashrcに設定しているので、`cd $EUSDIR`は思いつくと思います。

yurikadoi, 2015-11-22:
ご回答有難うございます。わかりました。

furushchev, 2016-07-13:
@yurikadoi If this issue is already solved, could you please close this issue?