euslisp/jskeus #307: 関節の角度を指定する方法

yurikadoi, 2015-11-18:
```
mech-user@test1-pc:/opt/ros/indigo/share/euslisp/jskeus/eus/models$ irteusgl h7-robot.l 
1.irteusgl$ setq *robot* (h7)
object#<h7-robot #X659e3c8 h7  0.0 0.0 0.0 / 0.0 0.0 0.0>
2.irteusgl$ s (list *robot*)
;; (make-irtviewer) executed
(#<h7-robot #X659e3c8 h7  0.0 0.0 0.0 / 0.0 0.0 0.0>)
3.irteusgl$ send *robot*  :larm :wrist-y :joint-angle 30
30
```
で、できることを今知りましたが、どこに書いてあるんでしょうか？

wkentaro, 2015-11-21:
http://euslisp-docs.readthedocs.org/en/latest/jskeus/jmanual/
に`:angle-vector`の説明があるのでその流れでソース見に行ってgrepしてると知れるのかもしれませんね。
僕も知らなかったので、もしjmanualに書いてないなら追加して下さい。

yurikadoi, 2015-11-22:
ソースを見に行ってgrepというのは具体的にどういうことか教えていただけますでしょうか？初学者なものですいません

wkentaro, 2015-11-22:
```
git clone https://github.com/euslisp/jskeus
cd jskeus
grep angle-vector
```

yurikadoi, 2015-11-23:
@wkentaro さん
ありがとうございます。わかりました。

furushchev, 2016-07-13:
@yurikadoi If this issue is already solved, could you please close this issue?