euslisp/jskeus #314: 関節の可動域を制限してIKをとく

snozawa, 2015-11-28:
関節の可動域を制限してIKをときたいことがあります（マージンをとっておく）。
だいぶ前から（rbrainのときに）していたことがあり、最近も別なコードでしていたことがあります。
理由はeuslispでIKをといたときに関節ぎりぎりだと、その後の実機での修正（st, imp, abcなど）で
可動域リミットにかかりやすくなってしまいます。
https://github.com/euslisp/jskeus/issues/196#issuecomment-95128551
にあるように、min-maxテーブルでマージンをとるのがちょっと大変そう

snozawa, 2015-12-21:
メモ：引数で与えたマージン分、:inverse-kinematics中で減じる。

mmurooka, 2017-07-31:
こちら古めのIssueですが，
min-maxテーブル関節の可動域を狭くしてIKを解きたいことが最近あったのですが，今もまだ難しいでしょうか．

snozawa, 2017-07-31:
そうですね、このissueをつくってからまだ検討を進められてません。
必要、と思ったことは何度かあります。
方法としては、単純にmin-maxテーブルを狭くしたものを外から与える、くらいしか現状思いつけてません。

mmurooka, 2017-08-01:
ありがとうございます．

mmurooka, 2017-09-21:
> 方法としては、単純にmin-maxテーブルを狭くしたものを外から与える、くらいしか現状思いつけてません。

で，決まった角度ぶんだけmin-maxテーブルを狭くする関数を作りました．

以下の2つのファイルを同じディレクトリにダウンロードして，
```
roseus min-max-table-offset-test-hrp2.l
(check-hrp2-wrist-table-offset)
```
すると，HRP2で右手首のmin-maxテーブルを5度狭くしてIKを解けます．
https://github.com/mmurooka/euslib/blob/murooka-branch/demo/murooka/wholebody_manipulation_planner/demo_robot_carry_object/euslisp/upstream/min-max-table-offset.l
https://github.com/mmurooka/euslib/blob/murooka-branch/demo/murooka/wholebody_manipulation_planner/demo_robot_carry_object/euslisp/upstream/min-max-table-offset-test-hrp2.l
これを実行したあとに，
````
gnuplot
plot "/tmp/table1-orig.dat" u 1:2 w lp, "" u 1:3 w lp
replot "/tmp/table2-orig.dat" u 2:1 w lp, "" u 3:1 w lp
replot "/tmp/table1-new.dat" u 1:2 w lp, "" u 1:3 w lp
replot "/tmp/table2-new.dat" u 2:1 w lp, "" u 3:1 w lp
````
すると以下のようなグラフが得られて正しく狭くなっていると思います．

[image]

最終的にはマクロにしてirtmodel.lにおいてもいいと思うのですがいかがでしょうか．
```
(with-min-max-table-offset *robot* (send *robot* :rarm :wrist-r) (send *robot* :rarm :wrist-p) 5.0
  (send *robot* :inverse-kinematics ...))
```
みたいな感じでしょうか．