euslisp/jskeus #317: euslispでソケット通信が簡単にできると聞いたんですが本当でしょうか？

Horimotsu, 2015-12-04:
@k-okada

Horimotsu, 2015-12-04:
なるほど、
https://github.com/euslisp/jskeus/blob/8ff7b08210bb5103c647830cf36274dee74f2116/irteus/irtutil.l#L127
をみてわかりました。
ターミナル１で
````
(eus-server)
```
として、もうひとつのターミナルで
```
$ (setq s (connect-server-until-success (unix::gethostname) 6666))
$ (format s "(+ 1 2)~%")
nil
$ (read s)
3
$ (format s "(objects (make-cube 100 100 100))~%")
nil
$ (format s "(setq *box* (make-cube 100 100 100))~%")
nil
```

@knorth55

Horimotsu, 2015-12-04:
```
1.irteusgl$ (connect-server-until-success "localhost" 6666)
;;; Can't connect server (localhost@6666-6685).
nil
```
となりますね。localhostではだめなのと、6666-6685というのがよくわかりませんでした。

k-okada, 2015-12-04:
動いてよかったですね．最後の部分が気になるので，
```
gethostip `hostname`
```
と
```
gethostip localhost
```
の結果を送ってもらいますか．

Horimotsu, 2015-12-04:
mech-user@test1-pc:~$ gethostip `hostname`
test1-pc 127.0.1.1 7F000101
mech-user@test1-pc:~$ gethostip localhost
localhost 127.0.0.1 7F000001

こんな感じになりました。
ホスト名だけ違う...?