euslisp/jskeus #318: documentation for inverse-kinematics in euslisp

k-okada, 2015-12-04:
```
searching "inverse kinematics", "IK", or "euslisp inverse kinematics" returns mostly issue pages. It would be nice to have a documentation page that explains how to use the function and its different input parameters.
```
from @wesleypchan

k-okada, 2015-12-04:
you can find function document at https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L319,
and find example codes, by selecting `code tag` at http://jsk-docs.readthedocs.org/en/latest/#gsc.tab=0&gsc.q=ik&gsc.sort=&gsc.ref=more%3Acodes

In japanese, there are theoretical background at 
http://euslisp.github.io/jskeus/jmanual-node152.html and exampel code at http://euslisp.github.io/jskeus/jmanual-node169.html

But, yes there are no good documentation for ik, for example did someone know where we can find documents for ':move-end-pos' and `:move-end-rot`?

I think IK is too basic feature for me and @snozawa so we do not realize what information is required for people, @wesleypchan, could you probide draft text for the ik documentation?You can leave blank  the information you need but do not know. For example

To solve inverse kinematics, you can use {...} method with { .... } keyword, 
there return values is { ... } etc...

wesleypchan, 2015-12-07:
Thank you for the references.

I've drafted up the text for an IK tutorial:

[ik-doc1.txt](https://github.com/euslisp/jskeus/files/53501/ik-doc1.txt)
[ik-doc2.txt](https://github.com/euslisp/jskeus/files/53502/ik-doc2.txt)
[ik-doc3.txt](https://github.com/euslisp/jskeus/files/53503/ik-doc3.txt)

I've separated it into three parts (github won't let me upload .l files so I've renamed them to .txt)