euslisp/jskeus #319: specifying linear trajectory in euslisp

k-okada, 2015-12-08:
```
ウェスリーです。

euslispで、ロボットの手を、ある初期位置から、あるゴール位置まで、直線に動かせる方法はなんでしょうか（添付図１，２）

今やっているのは、まず:inverse-kinematicを使って、ゴール位置を計算してから、angle vectorを実機を送るですが、結果は添付図３のように、直線にならず、隣にある机を当たってしまいます。

euslispでは直線に動かせる関数はありますか。

よろしくお願いします。
```

from @wesleypchan

k-okada, 2015-12-08:
you can find sample code at here -> https://gist.github.com/k-okada/3a151ef24415eb89d281

% gist is the best place to paste your code snippiest

k-okada, 2015-12-08:
sample code with `*ri*` -> https://gist.github.com/k-okada/c6c20f420c68893534ac

wesleypchan, 2015-12-08:
Thank you for the code. 

So I guess the solution is to create our own trajectory and write our own loop to move through each intermediate point.
Looking at the code in the link. if the robot is able to reach the goal point (that is, the IK is solvable for the goal position), but it is unable to solve the IK for one of the intermediate points, then it will stop in the middle of the trajectory, right?

So should we check and make sure that all the IK's for the intermediate points were solvable first before sending then angle vectors to *ri*?

k-okada, 2015-12-08:
yes, you're correct, if `:inverse-kinematics` fail to solve ik, it returns nil, so you can check them.