euslisp/jskeus #320: documentation for dynamic/balancing motion

ban-masa, 2015-12-09:
https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L623
にある　:move-centroid-on-foot が基本

ban-masa, 2015-12-09:
一番複雑なサンプルはhttps://github.com/euslisp/jskeus/blob/master/irteus/demo/walk-motion.l

snozawa, 2015-12-09:
:centroid で重心がでる．

snozawa, 2015-12-09:
:calc-zmp の引数で:calc-statics-p tにすると，静力学をときつつ，重心の投影点を計算してくれます．

ban-masa, 2015-12-09:
https://github.com/euslisp/jskeus/blob/master/irteus/irtdyna.l#L897
calc-statics-p がない気がします

snozawa, 2015-12-09:
すいません，
https://github.com/euslisp/jskeus/issues/320#issuecomment-163159192
の方法ではだめでした．
修正しますが，
```
(dotimes (i 2) (send *robot* :calc-zmp))
(send *robot* :calc-zmp)
```
してくれれば，`z= 0.0`の点での重心投影点をかえしてくれます．

ban-masa, 2015-12-09:
memo: https://github.com/euslisp/jskeus/issues/228, https://github.com/euslisp/jskeus/issues/319

k-okada, 2017-01-22:
`:centroid-on-foot` というメソッド．昔会った気がするけど、今は使っていないかな．その代わりに `:calc-zmp` がつかわれているのかな．