euslisp/jskeus #322: faceset の :update を消すと，順運動学が 45倍早くなる

s-noda, 2015-12-19:
eus の順運動学が遅すぎるきがして調べてみたのですが，
faceset クラスの `:update` という関数がボトルネックになっていることがわかりました．

以下テストコード
ランダムな関節角度に対して，全リンクの順運動学を１００回といて，計算速度を比較しています．
はじめに普通に計算して，次に， faceset:update を何もしないよううわがいて再度同じ計算をします．
```
(require "package://hrpsys_ros_bridge_tutorials/euslisp/hrp2jsk-interface.l")

(setq *robot* (hrp2jsk))
(objects (list *robot*))

(defun random-fk-test
  (&rest args)
  (mapcar '(lambda (j) (send j :joint-angle
                             (- (+ (* (random 1.0) (- (send j :max-angle) (send j :min-angle)\
))
                                   (send j :min-angle))
                                (send j :joint-angle))
                             :relative t))
          (send *robot* :joint-list))
  (let* ((tm (bench2 (send-all (send *robot* :links) :worldcoords))))
    (send *viewer* :draw-objects)
    tm))

(defun remove-self-update
  nil
  (defmethod FaceSet
    (:update nil nil))                                      
  )

(let* ((org-tm (apply '+ (mapcar 'random-fk-test (make-list 100))))
       (fast-tm (progn (remove-self-update)
                       (apply '+ (mapcar 'random-fk-test (make-list 100))))))
  (warning-message 1 "~A/~A = ~A~%" org-tm fast-tm (/ org-tm fast-tm)))
```

僕の手元での実行結果は
`0.602862/0.013417 = 44.9327 `
で， faceset の update を行わないと， 45倍早くなります．

faceset:update の中身は
https://github.com/euslisp/EusLisp/blob/master/lisp/geo/geobody.l#L25
で， face の頂点を更新しているようです．

そこで，ここから先自信がない，というか僕のしらない深い理由があるのではないかと思っての質問なのですが，face の頂点を更新する必要はあるのでしょうか？
irtviewer の表示は問題ないようです．
衝突は pqp を使う分には正しく動いているようで， faceset の更新を行う必要はないのではないかと思います．
困るのは， link:draw-on した時に，全然違う姿勢が見えてしまうことぐらい？

k-okada, 2015-12-19:
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L694
analysis-level で:body と:coordsを選択できて，:coordsだとkinematicsレベルだけの更新になって
faceはこうしんされないようにしているつもりですがどうでしょうか．

最近のロボットはcolladaからつくっているとfaceは使ってなくて，openglで直接表示していんだろうけど，
昔の手で作ったロボットだとirtviewerの表示は変になりますね．ただ，openglで表示しているロボットは
faceはconvex hullとかbounding boxのつもりでしたが，そうでもなくなっているのかな．

s-noda, 2015-12-19:
> analysis-level で:body と:coordsを選択できて，:coordsだとkinematicsレベルだけの更新になって

なるほど！　重要な知識をありがとうございます．
確かに，上のコードに `(send-all (send *robot* :links) :set-val 'analysis-level :coords)` を追加すると，差はほとんどなくなりました．

> 最近のロボットはcolladaからつくっているとfaceは使ってなくて，openglで直接表示していんだろうけど，

なるほど，では，ないと困るプログラムもありそうですね．

僕は，fk が早くなってくれれば問題ないので， `analysis-level` を使っていこうと思います．

closed 2015-12-19 by s-noda

k-okada, 2015-12-21:
どこかにドキュメント書きましょうか．

s-noda, 2015-12-21:
> どこかにドキュメント書きましょうか．

どこにしましょう．

irtmodel.l の内容なので，jmanual はおかしいですよね．

https://github.com/euslisp/jskeus/blob/master/doc/irtmodel.tex#L5

とかに，新しい subsection を作る感じがいいでしょうか．

k-okada, 2015-12-23:
そうですね．それで書いてみて下さい．