euslisp/jskeus #329: :calc-zmpの挙動が時々不安定になる

garaemon, 2015-12-28:
モデルはhrp2jskntを利用.

以下のようにzmpの値が突然大きくなることがあります。
プロットしているのは:calc-zmpのx, yの値です

[image]

テストコード
https://gist.github.com/garaemon/104d2dee4736754992ab

garaemon, 2015-12-28:
:dt の値によって挙動が変わってくるように見えます

snozawa, 2015-12-28:
robotは何ですか？

snozawa, 2015-12-28:
ごめん、hrp2jskntでしたね。

snozawa, 2015-12-28:
多分calc-zmpに入力される速度加速度が不連続になっています。
まず補間がlinear-interpolatorだとその変化点で加速度が過大になるので、
minjerk-interpolatorにしてください。
次は、:move-centroid側の問題っぽいのですが、:min-loop 2を:min-loop 3にしてみてくれますでしょうか。

garaemon, 2015-12-29:
minjerk + :min-loop 3で挙動が良くなりました

[image]

closed 2015-12-29 by garaemon