euslisp/jskeus #334: :calc-jacobian-from-link-listで計算されるヤコビアンの関節対応の順番

k-kimura, 2016-01-07:
irtmodel.lで定義されている:calc-jacobian-from-link-list
(https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L1067-#L1151)
の計算で返されるヤコビアン(6自由度*関節数)の列要素に対応する関節の順番についてお聞きしたいです．

例として\*hrp2jsk*の場合
```
irteusgl$ (setq *robot* *hrp2jsk*)
#<hrp2jsk-robot #X8809c70 HRP2JSK  0.0 0.0 0.0 / 0.0 0.0 0.0>
irteusgl$ (setq *sample-rarm-jacobian*                                                                                    
      (send *robot* :calc-jacobian-from-link-list (send *robot* :link-list (send *robot* :rarm :end-coords :parent))          
            :move-target (send *robot* :rarm :end-coords)                                                                    
            :translation-axis t                                                                                              
            :rotation-axis t))
#2f((0.130005 -0.101614 -0.185651 -0.013844 -0.009693 -0.043412 -0.002935 -0.016643 -4.263256e-17) (0.092094 0.031587 0.035677 0.619109 0.205457 1.204926e-18 0.030215 0.171357 1.183291e-33) (0.318766 -0.457443 -0.617899 0.009693 -0.013844 -0.420202 -0.016643 0.002935 -0.174) (-0.894052 -0.142244 -0.142244 0.573576 -0.819152 2.775558e-17 -0.984808 0.173648 2.775558e-17) (-0.17101 0.984808 0.984808 0.0 0.0 1.0 0.0 -1.387779e-17 1.0) (0.414037 0.099601 0.099601 0.819152 0.573576 0.0 0.173648 0.984808 0.0))
irteusgl$ (array-dimensions *sample-rarm-jacobian*)
(6 9)
```
のように
:rarmのエンドエフェクタの位置姿勢速度(6自由度)を行要素，
rootリンクから:rarmの手先リンクRARM_LINK6までの関節数(RARM_JOINT7を除く9自由度)を列要素，
とするヤコビアンが返されますが，
列方向に対する関節の順序のようなものはeus上で調べることができますでしょうか？
(ヤコビアンに掛ける関節角速度列の順番を知りたいです．)

ちなみに
```
irteusgl$(send *robot* :rarm :joint-list)
(#<rotational-joint #Xa857918 RARM_JOINT0> #<rotational-joint #Xa85bbd8 RARM_JOINT1> #<rotational-joint #Xa85baa0 RARM_JOINT2> #<rotational-joint #Xa85b968 RARM_JOINT3> #<rotational-joint #X8853fe8 RARM_JOINT4> #<rotational-joint #Xa841de8 RARM_JOINT5> #<rotational-joint #Xa841cb0 RARM_JOINT6> #<rotational-joint #Xa841b78 RARM_JOINT7>)
```
とすると:rarm系のすべての関節の順序リストが返ってきますが，
ヤコビアン計算のほうでは含まれているCHESTの関節(pitch,yawの2自由度)
が含まれない形となっています．

snozawa, 2016-01-07:
このあたりのコードは、関節順番は全てlink-listの順番になっています。
このメソッドでは
```
(send-all (send *robot* :link-list (send *robot* :rarm :end-coords :parent)) :joint)
```
になります。

ちなみに、複数分岐のあるものもirteusのヤコビアン計算はサポートしています。
双腕IKの中で登場するヤコビアンの例では
```
(setq *sample-jacobian*                                                                                    
      (send *robot* :calc-jacobian-from-link-list
            (mapcar #'(lambda (l) (send *robot* :link-list (send *robot* l :end-coords :parent))) '(:rarm :larm))
            :move-target (list (send *robot* :rarm :end-coords) (send *robot* :larm :end-coords))
            :translation-axis (list t t) :rotation-axis (list t t)))
(format-array *sample-jacobian*)
```
のようにして表示されます。
ヤコビアンの形が、胴体部分が２列あって、その隣がブロック対角行列になってるのがわかると思います。
この関節順序は
```
(setq link-list (mapcar #'(lambda (l) (send *robot* :link-list (send *robot* l :end-coords :parent))) '(:rarm :larm))))
(send-all (send *robot* :calc-union-link-list link-list) :joint)
```
のようにして得られます。
:calc-union-link-listは、これ系のメソッドの中でよばれてたりします。

複数分岐があってもなくても:calc-union-link-listして:jointをとってくれば間違いはないです。

k-kimura, 2016-01-07:
ありがとうございます．
基本的には関節順序はlink-listの順から定義されているのですね．

双腕分岐型の関節順序もヤコビアンの列からみて
(共通のCHESTリンク2列) + (右腕リンク7列) + (左腕リンク7列)
になっているのが確認できました．

:calc-union-link-listで分岐型のlink-listも統一できることは知らなかったので覚えておきます．

ちなみに些細な確認ですが
```
;; 左腕のエンドエフェクタx成分に1[N]与えたときの関節トルク列[Nm]
l$ (transform (transpose *sample-jacobian*) (float-vector 0 0 0 0 0 0 1 0 0 0 0 0))
#f(-0.130005 -0.101614 0.0 0.0 0.0 0.0 0.0 0.0 0.0 -0.185651 0.013844 0.009693 -0.043412 0.002935 0.016643 -4.263256e-17)
```
で出てくる関節トルク列[Nm]も
```
(send-all (send *robot* :calc-union-link-list link-list) :joint)
```
の関節順，という理解で正しいでしょうか．
(右腕系の関節トルクが0[Nm]となっているので合ってはいそうです)

k-okada, 2016-01-08:
疑問点だった部分を
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L1090
のコメントに追加しておいて下さい． > @k-kimura

snozawa, 2016-01-08:
> の関節順，という理解で正しいでしょうか．
> (右腕系の関節トルクが0[Nm]となっているので合ってはいそうです)

はい、あってます。

k-kimura, 2016-01-08:
> のコメントに追加しておいて下さい． > @k-kimura

#335 で対応しました．

closed 2016-01-08 by snozawa