euslisp/jskeus #348: 首を動かしてものを見回すIK

YuOhara, 2016-03-11:
その場で覗き込んで、様々な角度からものを見るためのIKを解きたいです。
以下のようにfullbody_ikをしてもうまくいきませんでした。

```
(defun move-head (head-tc &key (draw? nil) (translation-axis (list t t t)) (rotation-axis (list t t t)) (stop 100))
  (let
      ((tc (list
            head-tc
            (send *robot* :rleg :end-coords :copy-worldcoords)
            (send *robot* :lleg :end-coords :copy-worldcoords)))
       )
    (with-move-target-link-list
     (mt ll *robot* (list :head :rleg :lleg))
     (send *robot*
           :fullbody-inverse-kinematics tc
           :move-target mt :link-list ll
           :additional-weight-list
           (list (list (send *robot* :rleg :toe-p :child-link) 0)
                 (list (send *robot* :lleg :toe-p :child-link) 0)
                 (list (send *robot* :torso :waist-p :child-link) 0.1)
                 (list (send *robot* :torso :waist-y :child-link) 0.2)
                 )
           :root-link-virtual-joint-weight (float-vector 0.1 0.1 0.1 0.1 0.5 0.5)
           :target-centroid-pos (apply #'midpoint 0.5 (send *robot* :legs :end-coords :worldpos))
           :translation-axis translation-axis
           :rotation-axis rotation-axis
           :max (float-vector  500  500  0  20  20  10) ;; for hrp2
           :min (float-vector  -500  -500  -200  -20  -20  -10) ;; for hrp2
           :stop stop
           :debug-view (if draw? :no-message nil)
           )
     ))
  )
(defun test ()
  (load "package://hrpsys_ros_bridge_tutorials/euslisp/hrp2jsknts-interface.l")
  (hrp2jsknts-init)
  (setq *robot* *hrp2jsknts*)
  (objects *robot*)
  (move-head (send (send (send *robot* :head :end-coords) :copy-worldcoords) :translate (float-vector 0 10 0)) :rotation-axis (list t t t))
  )

```

k-okada, 2016-03-11:
Please look at this
https://github.com/euslisp/jskeus/pull/157

mmurooka, 2016-03-11:
（内部向けですが）以下のコマンドで画像のような姿勢が作れるので参考にしてみてください．

```
roseus
(load "~/prog/euslib/demo/nozawa/ik/test-arm-head.l")
(test-look-at-with-arm-strictly-directioned)
```

[image]

YuOhara, 2016-03-11:
ありがとうございます。参考にします。

k-okada, 2016-03-12:
jskeus に samplerobot でサンプルプログラムを入れて下さい > @mmurooka

mmurooka, 2016-03-12:
サンプルを追加しました．