euslisp/jskeus #374: complie of irteus/*.l does not run after compile_*.log is generated.

snozawa, 2016-05-08:
After https://github.com/euslisp/jskeus/pull/373, build behaviour changed because of change of build dependency.
Complie of irteus/xx.l does not run after compile_xx.log is generated.

- Reproduce the problem
```
$ cd ~/ros/indigo_parent/src/Euslisp/jskeus/irteus
$ touch irtmodel.l
$ make
make: Nothing to be done for `all'.
```
The same problem occurs when i run `catkin bt` in jskeus directory.
If I remove compile_xx.log file, compile of irteus/xx.l starts.

- Expected behaviour   
compile_irtg.l is called invoked by irtmodel.l time-stamp update.

k-okada, 2016-05-08:
sorry, check if #375 fixes

k-okada, 2016-05-08:
see https://github.com/euslisp/EusLisp/pull/177 for euslisp case

snozawa, 2016-05-08:
Thanks!
BTW, in the original version, 
```
$(IRTEUSOBJS):
	$(BINDIR)/eus2$(ESFX) "(setq *objdir* \"$(INSTALLOBJDIR)/\")" < ./compile_irt.l > /dev/null
```
dependencies are not written explicitly, but it worked correctly.
Which part specifies dependency?

k-okada, 2016-05-09:
see
https://github.com/euslisp/EusLisp/blob/master/lisp/Makefile.generic2#L49
and
https://github.com/euslisp/EusLisp/blob/master/lisp/Makefile.generic2#L442

PR #375 (2016-05-12): compile_*.log depends on .l files
This is regresson on #373.

Closes #374
closed 2016-05-12 by k-okada