euslisp/jskeus #381: calc-zmp でanalysis-level をセットしている？

k-okada, 2016-06-14:
@snozawa bufferで毎回float-vector作らないようにはしていそうだが，@terasawaによると0.03秒かかっているらしい．とりあえずベンチマークのソフトを作って下さい．> @terasawa
a

terasawa, 2016-06-14:
私の環境では
```
(progn                                                                                                            
  (send *robot* :fix-leg-to-coords (make-coords))                                                                              
  (bench (send *robot* :calc-zmp))                                                                                             
  )
```
をすると何度やっても0.03秒以上はかかります。普通はこうはならないのでしょうか？

ちなみに私は最適化で:calc-zmpを計算する際に必要となる速度や加速度を、bsplineの真値からではなく３つの関節角度の差分とdtから計算しています。
（毎回２つ前の姿勢にしてからcalc-zmpを呼び、次に１つ前の姿勢にしてからcalc-zmpを呼び,
最後に求めたい姿勢にしてcalc-zmpを呼んでいて、呼ぶ際に毎回姿勢は変化しているのでテストではfix-leg-to-coordsを呼ぶようにしてみました）

この様子だと１つの時刻のzmpを計算するときは毎回0.1秒はかかっている計算になってしまっています。

そういうこともあって以前 #331 のissueを立てました（結局内部でかなり大変そうなことをやっていたので自分は諦めたのですが…）

snozawa, 2016-06-14:
こちらの手元だと、5かいやってみて
```
92.irteusgl$ (dotimes (i 5)                                                                                                                                              
  (send *robot* :fix-leg-to-coords (make-coords))                                                                                                                        
  (bench (send *robot* :calc-zmp)))
;; time -> 0.001139[s]
;; time -> 0.000787[s]
;; time -> 0.000788[s]
;; time -> 0.000794[s]
;; time -> 0.001093[s]
```
で、１ケタ２ケタ違いそうです。
それで、上記の時間ならEus実装ならこんなものかなあという印象です。
使うPCをかえてみられるかな？
（あと、対象としてるロボットはJAXONであってる？）

snozawa, 2016-06-14:
> この様子だと１つの時刻のzmpを計算するときは毎回0.1秒はかかっている計算になってしまっています。

これはなんでだろう？
最初は２時刻分前の情報がいるけど、それ以降は１こずつ時刻がずれてくだけなので、
全時刻でk = Kだとすると、K+2のcalc-zmp計算だから (K+2)*0.03の計算時間になるので、
毎回0.1秒（ K*0.1）とはちがうんじゃないかな。

terasawa, 2016-06-15:
> 使うPCをかえてみられるかな？
> （あと、対象としてるロボットはJAXONであってる？）

変えてみたのですが、僕が持っている３つのPCの最高速はi7-6700K/32GB/GTX970/mini-ITXで以下のようになっています。対象ロボットはJAXON(正確にはJAXON_RED)です。

```
$ (dotimes (i 5) (send *robot* :fix-leg-to-coords (make-coords)) (bench (send *robot* :calc-zmp)))
;; time -> 0.027418[s]
;; time -> 0.038357[s]
;; time -> 0.037388[s]
;; time -> 0.027042[s]
;; time -> 0.026677[s]
```

> 最初は２時刻分前の情報がいるけど、それ以降は１こずつ時刻がずれてくだけなので、
> 全時刻でk = Kだとすると、K+2のcalc-zmp計算だから (K+2)*0.03の計算時間になるので、
> 毎回0.1秒とはちがうんじゃないかな。

これですが、いつもZMPの不等式条件はぎりぎりのところで最適化のループが進んでいくので、できるだけ正確な値を与えたほうが良いかなと考えています。現状のZMPの不等式条件としては動作を適当に離散化してそのときのZMPが支持多角形の外に出ないようにという条件になっていて、（もともとロードパターンで送ることを想定していたので）いつもその姿勢でのZMPを計算する際に4ms前と2ms前の姿勢を一回呼んでいます。
ですので、このやり方が妥当かどうかはおいておいたとして現状は１個のZMPを計算する計算時間は3*0.03になっています。

k-okada, 2016-06-15:
(load "package://euslisp/jskeus/eus/irtmodel.l")
溶かしていると，コンパイルされていないので，遅くなるけど大丈夫かな？
サンプルプログラム全部pasteしてみよう

terasawa, 2016-06-15:
話が２つあって、１点目はそもそも:calc-zmpが遅そうという話で、サンプルをすべてペーストするとこんなかんじです

```
terasawa@terasawa-Desktop-tissa:~/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/euslisp (master)$ roseus jaxon_red-interface.l 
configuring by "/home/terasawa/ros/indigo_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql 
;; extending gcstack 0x5e6e580[16374] --> 0x6144110[32748] top=3962
irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb irtgl irtviewer                                                                                                     
EusLisp 9.19(f86cfee 1.0.12) for Linux64 created on terasawa-desktop2(Mon May 9 15:44:58 JST 2016)
roseus ;; loading roseus("173965c") on euslisp((9.19 terasawa-desktop2 Mon May 9 15:44:58 JST 2016 f86cfee 1.0.12))            
eustf roseus_c_util 
;; extending gcstack 0x6144110[32738] --> 0x6d51ad0[65476] top=7d9f
1.irteusgl$ jaxon_red
#<jaxon_red-robot #X6ae7c70 JAXON_RED  0.0 0.0 0.0 / 0.0 0.0 0.0>
2.irteusgl$ dotimes (i 5) (send *jaxon_red* :fix-leg-to-coords (make-coords)) (bench (send *jaxon_red* :calc-zmp))
;; time -> 0.02546[s]
;; time -> 0.025285[s]
;; time -> 0.025217[s]
;; time -> 0.024754[s]
;; time -> 0.023994[s]
nil
```

この１点目については特に変なものをloadしているわけではないので単に遅いんじゃないかと思っています。

で、２点目は私もだいぶ思い当たる節があって、上のサンプルではなく私のプログラムに関しての話になりますが(load "package://euslisp/jskeus/eus/irtmodel.l")していて、別issueのほうが良いかもしれませんが、一応このissueに書くと、
私のプログラムでは最適化の際にルートリンクの６自由度も含めていて、 
```
(with-append-root-joint
  (link-list-with-robot-6dof *robot* (list (cdr (send *robot* :links)))
                             :joint-class 6dof-joint)
  ;; ここで最適化計算を走らせる
  )
```
のようになっているのですが、ルートリンクに仮想リンクを足した状態だとforward-kinematicsを呼んだ時に仮想リンクが親リンクを参照しに行ってエラーをはいてしまうので、:forward-all-kinematics（と:inverse-dynamics）を以下のように上書きして使っています。
```
-   (when parent-link
+   (when (and parent-link (not (eq (send parent-link :name) 'virtual-link)))
```
そのため、irtmodel.lやirtdyna.lの中にある関数を使うときに遅くなっているかもしれません。

ちなみにそのファイルは具体的には以下のようなものです。
```
(require "package://jskeus/irteus/irtmodel.l")
(require "package://jskeus/irteus/irtdyna.l")

(defmethod bodyset-link
  (:forward-all-kinematics
   (&key (debug-view nil)
         ;; buffers for calculation
         (tmp-va (float-vector 0 0 0)))
   (if debug-view (format t ";; forward-all-kinematics link = ~A~%" (send self :name)))
   (when (and parent-link (not (eq (send parent-link :name) 'virtual-link)))
     (let* ((paxis (case (joint . axis)
                         (:x #f(1 0 0)) (:y #f(0 1 0)) (:z #f(0 0 1))
                         (:xx #f(1 0 0)) (:yy #f(0 1 0)) (:zz #f(0 0 1))
                         (:-x #f(-1 0 0)) (:-y #f(0 -1 0)) (:-z #f(0 0 -1))
                         (t (joint . axis))))
            (ax (normalize-vector (send (send (send parent-link :copy-worldcoords) :transform
                                              (joint . default-coords))
                                        :rotate-vector paxis tmp-va)
                                  (send self :get :angular-velocity-jacobian))) ;; []
            (svj (send joint :calc-spacial-velocity-jacobian ax tmp-va
                       (send self :get :spacial-velocity-jacobian)))
            (avj (send joint :calc-angular-velocity-jacobian ax
                       (send self :get :angular-velocity-jacobian))))
       (setq angular-velocity ;; [rad/s]
             (v+ (send parent-link :angular-velocity)
                 (scale (send joint :joint-velocity) avj tmp-va) angular-velocity)
[... 39 lines cut ...]
                                  (v* angular-acceleration c tmp-vc) force) force)
                     (v* angular-velocity momentum tmp-vc) force) ;; [N]                                                                                                                               
           moment (v+ (v+ (v+ (scale w (v* c spacial-acceleration tmp-vc) moment)
                              (transform I angular-acceleration tmp-vc) moment)
                          (v* spacial-velocity momentum tmp-vc) moment)
                      (v* angular-velocity angular-momentum tmp-vc) moment)) ;; [N/m]                                                                                                                  
     ;; use ext-force and ext-moment                                                                                                                                                                   
[... cut ...]
           :tmp-va tmp-va :tmp-vb tmp-vb :tmp-vc tmp-vc
           :tmp-ma tmp-ma :tmp-mb tmp-mb :tmp-mc tmp-mc :tmp-md tmp-md)
     (setq force (v+ force (send child :force) force)
           moment (v+ moment (send child :moment) moment)))
   (when (and joint parent-link (not (eq (send parent-link :name) 'virtual-link)))
     (send joint :joint-torque
           (+ (v. (send self :get :spacial-velocity-jacobian) force)
              (v. (send self :get :angular-velocity-jacobian) moment))))
   )
  )
```

snozawa, 2016-06-15:
２つめの問題は、PRを作成してみてください。

１つめの問題は先ほどオフラインで@terasawaくんとanalysis-levelの問題ということを確認しました。
その後:forward-all-kinematicsの前後でanalysis-levelをかえてみて、ここがanalysis-levelがbodyだと遅くなる要因になってるというところまで突き止めました。
ましたが、どこでanalysis-level bodyが効いてくる計算になってるかはまだ未確認です（そうなってないはず、なコード）

snozawa, 2016-06-15:
> ましたが、どこでanalysis-level bodyが効いてくる計算になってるかはまだ未確認です（そうなってないはず、なコード）

cascaded-coordsの:worldposを使っていて、
https://github.com/euslisp/jskeus/blob/master/irteus/irtdyna.l#L521
:worldposは中で:worldcoordsをよんでるので、ここが原因なきがしました。

@terasawa君
１つめの問題も、:calc-torque-from-vel-accあたりでanalysis-levelを:coordsにして:bodyに戻す変更をしてみてください。

terasawa, 2016-06-15:
ありがとうございました。
PR( #382 #383 )を作成いたしました。

snozawa, 2016-06-20:
実行時間に関してオフラインで確認したところ
- 最初の@snozawaと@terasawaとで実行時間がちがうのは、ロボットがちがうからでした（sample-robotとjaxon_red)。jaxon_redなら確かに少し遅くなりました。
- :calc-torqueの中の:forward-all-kinematicsメソッドで:worldposをよんだりしてfkがよばれてしまうので、analysis-levelをかえたら早くなりました(jaxon_redで確認)。

ということで、遅いのは
https://github.com/euslisp/jskeus/pull/383
で治ると思います。