euslisp/jskeus #384: let文で宣言されたbodyクラスのインスタンスがおかしい

pazeshun, 2016-06-21:
以下のように、グローバルに宣言されたインスタンスと、ローカルに宣言されたインスタンスとで結果が違うことがあります。
```
$ irteusgl
configuring by "/opt/ros/indigo/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x4a17de0[16374] --> 0x4e73980[32748] top=3c7e
irtgl irtviewer 
EusLisp 9.19( 1.0.12) for Linux64 created on ip-172-31-15-39(Thu May 5 01:33:05 PDT 2016)
1.irteusgl$ setq cube (make-cube 100 100 100)
#<body #X4cd9130 (:cube 100.0 100.0 100.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>
2.irteusgl$ send cube :rotate 0.3 :x
#<body #X4cd9130 (:cube 100.0 100.0 100.0) 0.0 0.0 0.0 / 0.0 0.0 0.3>
3.irteusgl$ send cube :vertices
(#f(50.0 62.5428 -32.9908) #f(50.0 32.9908 62.5428) #f(50.0 -32.9908 -62.5428) #f(50.0 -62.5428 32.9908) #f(-50.0 62.5428 -32.9908) #f(-50.0 -32.9908 -62.5428) #f(-50.0 -62.5428 32.9908) #f(-50.0 32.9908 62.5428))
4.irteusgl$ let (b) (setq b (make-cube 100 100 100)) (send b :rotate 0.3 :x) (send b :vertices)
(#f(50.0 50.0 -50.0) #f(50.0 50.0 50.0) #f(50.0 -50.0 -50.0) #f(50.0 -50.0 50.0) #f(-50.0 50.0 -50.0) #f(-50.0 -50.0 -50.0) #f(-50.0 -50.0 50.0) #f(-50.0 50.0 50.0))
```
これは、どのように理解すればよいのでしょうか。
@wkentaro も似たような問題にあたっているようです。

pazeshun, 2016-06-21:
なお、インタプリタの表示を見ると、違いがないように見えます。
```
5.irteusgl$ cube
#<body #X4cd9130 (:cube 100.0 100.0 100.0) 0.0 0.0 0.0 / 0.0 0.0 0.3>
6.irteusgl$ let (b) (setq b (make-cube 100 100 100)) (send b :rotate 0.3 :x) b
#<body #X4e509d8 (:cube 100.0 100.0 100.0) 0.0 0.0 0.0 / 0.0 0.0 0.3>
```

k-okada, 2016-06-22:
```
$ cat hoge.l 
#!/usr/bin/env irteusgl

(print
 (progn
   (setq b (make-cube 100 100 100))
   (send b :rotate 0.3 :x)
   (send b :vertices)))
(print
 (progn
   (setq b (make-cube 100 100 100))
   (send b :rotate 0.3 :x)
   (send b :worldcoords)
   (send b :vertices)))
(print (let (b)
         (setq b (make-cube 100 100 100))
         (send b :rotate 0.3 :x) 
         (send b :vertices)))
(print (let (b)
         (setq b (make-cube 100 100 100))
         (send b :rotate 0.3 :x) 
         (send b :worldcoords)
         (send b :vertices)))
```

pazeshun, 2016-06-22:
原因は、cubeを回転/移動させた際に、let文内だと頂点情報が更新されていなかったことでした。
`(send b :worldcoords)`をすることで更新されました。

closed 2016-06-22 by pazeshun

knorth55, 2016-06-22:
```
2.irteusgl$ send cube :methods
(:translate-vertices :rotate-vertices :magnify :euler :perimeter :volume :centroid :world-centroid :area :extream-point :length :supporting-faces :possibly-interfering-faces :possibly-interfering-edges :intersect-face :intersectp :intersectp2 :insidep :evert :set-convexp :get-face :primitive-body-p :primitive-bodies :csg :copy-csg :body-type :creation-form :prin1 :init :replace-shape :+ :- :* :primitive-groups :constraint :contact-vertices :contact-edges :possibly-contacting-vertices :possibly-contacting-edges :possibly-contacting-faces :update :vertices :faces :face :all-edges :edges :edge :vertex :box :color :reflectance :diffusion :holes :visible-faces :visible-edges :contour-edges :non-contour-edges :common-box :newbox :reset-vertices :translate-vertices :rotate-vertices :magnify :faces-intersect-with-point-vector :distance :init :set-color :draw-on :paste-texture-to-face :parent :descendants :inheritance :leaves :assoc :dissoc :clear-assoc :obey :disobey :newcoords :changed :worldrot :worldpos :worldcoords :copy-worldcoords :update :parentcoords :transform-vector :rotate-vector :inverse-transform-vector :inverse-transformation :transformation :transform :move-to :rotate-with-matrix :rotate :orient-with-matrix :orient :parent-vector :parent-orientation :manager :init :move-coords :inverse-rotate-vector :make-pqpmodel :dimension :rot :pos :x-axis :y-axis :z-axis :name :newcoords :replace-rot :replace-pos :replace-coords :copy-rot :copy-pos :copy-coords :coords :worldrot :worldpos :worldcoords :copy-worldcoords :parentcoords :changed :reset-coords :move-to :rotate-vector :transform-vector :inverse-transform-vector :inverse-transformation :transformation :transform :rotate-with-matrix :rotate :orient-with-matrix :orient :parent-vector :parent-orientation :translate :locate :scale :euler :euler-angle :rpy :rpy-angle :rotation-angle :4x4 :prin1 :create :init :axis :difference-position :difference-rotation :inverse-rotate-vector :vertices :draw-on :plist :get :put :name :remprop :prin1 :prin1 :warning :error :slots :methods :super :get-val :set-val)
```
methodsをみてみたのですが:verticesが二つあるのですがこれは正しいのでしょうか？

k-okada, 2016-06-22:
親クラスとこクラスで定義されているかな．

knorth55, 2016-06-22:
その場合はoverrideされるという解釈で間違いないですか？