euslisp/jskeus #389: interpolator

k-okada, 2016-07-17:
irtuesにinterpolatorがあるといいとおもうんだけど， > @snozawa 
1) https://github.com/jsk-ros-pkg/euslib/commits/master/rbrain/virtual_interpolator.l はつかっているかな？
2) irteusないでinterpolation しているのは，https://github.com/euslisp/jskeus/blob/master/irteus/irtdyna.l#L1794-L1803 ぐらい，というのは正しい？

snozawa, 2016-07-20:
> 2) irteusないでinterpolation しているのは，https://github.com/euslisp/jskeus/blob/master/irteus/irtdyna.l#L1794-L1803 ぐらい，というのは正しい？

はい、そうです。足軌道・こし軌道を出力するところで使っています。

> 1) https://github.com/jsk-ros-pkg/euslib/commits/master/rbrain/virtual_interpolator.l はつかっているかな？

ぼくは使っていて、他の学生にも勧めたことがあります。
テスト（というかサンプル）は
https://github.com/jsk-ros-pkg/euslib/blob/master/tests/test-virtual-interpolator.l
にあります。
virtual_interpolator.l自体はrbrainに依存せず使えるので、
irteusのみを使う内部の人であればloadすれば使える形になってましたので、
irteus移行は今までしてませんでした。

また、個人的にはこのレイヤでは
- オフライン補間：経由点列があらかじめすべてもとまっていて、一期に補間列も求める（pos-list-interpolation関数）
- オンライン補間：経由点がオンラインで追加されて、瞬時の補間計算結果を取得して使う、とループで回す（:pass-timeメソッドを使う）

のうち前者しか使ったことがありません。
（後者は、hrpsysのseqでやっていることに相当します）

k-okada, 2016-07-20:
#391 見て下さい．

snozawa, 2017-01-26:
Solved via https://github.com/euslisp/jskeus/pull/391

closed 2017-01-26 by snozawa