euslisp/jskeus #398: use-baseのとき、ロボットの移動に対してヤコビアン行列が変わらない問題

chiwunau, 2016-10-30:
手先が力を受ける時に関節負荷超えない姿勢を取ることのために関節トルクの勾配を計算していて、ヤコビアンの角度に対する微分の数値計算のところで以下の問題がでてきます。

マニピュレータの関節角を変えたらヤコビアンも変わって、これに関しては問題ないですが、Baseの移動をいれたら以下のコードではbaseの仮想リング?の角度を変えてもヤコビアンが変わらないのが問題です。
つまりdiff-jaco1はゼロ行列ではないが、diff-jaco2はゼロ行列になる。

原因はmove-targetの位置はlink-listからロボットのベース相対に計算されるからだと思いますが、地面に固定した仮想リングの設定、或いは他の解決策はあるのでしょうか

```                                                                                        
(load "package://pr2eus/pr2-interface.l)
(pr2)
(setq link-list (send *pr2* :link-list (send *pr2* :rarm :end-coords)))
(send *pr2* :reset-pose)

(with-append-root-joint
 (ll *pr2* link-list
     :joint-class omniwheel-joint
     :joint-args (list :min #f(-500 -500 -90) :max #f(500 500 90)))
     (send (caar ll) :joint :joint-angle #f(5 5 5))

     (send *pr2* :rarm :shoulder-p :joint-angle 0)
     (setq jaco1 (send *pr2* :calc-jacobian-from-link-list ll :move-target (send *pr2* :rarm :end-coords) :rotation-axis t :translation-axis t))
     (send *pr2* :rarm :shoulder-p :joint-angle 10)
     (setq jaco2 (send *pr2* :calc-jacobian-from-link-list ll :move-target (send *pr2* :rarm :end-coords) :rotation-axis t :translation-axis t))
     (setq diff-jaco1 (m- jaco2 jaco1))

     (send (caar ll) :joint :joint-angle #f(0 0 0))
     (setq jaco1 (send *pr2* :calc-jacobian-from-link-list ll :move-target (send *pr2* :rarm :end-coords) :rotation-axis t :translation-axis t))
     (send (caar ll) :joint :joint-angle #f(100 0 0))
     (setq jaco2 (send *pr2* :calc-jacobian-from-link-list ll :move-target (send *pr2* :rarm :end-coords) :rotation-axis t :translation-axis t))
     (setq diff-jaco2 (m- jaco2 jaco1))
)
```

snozawa, 2016-10-31:
これは
https://github.com/euslisp/jskeus/issues/387
と同じ問題でしょうか？

chiwunau, 2016-10-31:
二足ロボットは詳しくないですが、２つの問題の原因は同じように気がします。
なにか一時的な解決策などありますか。。

chiwunau, 2016-10-31:
よくよく考えると、```(send (caar ll) :joint :joint-angle #f(100 0 0))```の並進運動に対してヤコビアン行列が変わらないほうが正しくて、#f(0 0 50)にしたらちゃんと変わっていました。
もうちょっと考えてみます。

#387 
二足歩行ロボットに詳しくないのでよくわからないですが、追加した仮想リングが世界座標系の中に固定ではなくロボットといっしょうに動いているのが問題だと思っていました。