euslisp/jskeus #399: bounding box を irtviewerで表示したい

h-kamada, 2016-11-09:
```
$ (setq b (make-bounding-box (list #f(0 0 0) #f(100 100 100))))
```
としてつくったbounding boxをobjectsすると
```
$ (objects (list b))
opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :worldcoords in (objects (list b))
```
となります。どうしたらいいでしょうか？

h-kamada, 2016-11-09:
自己解決しました！！！
```
(objects (list (send b :body)))
```
でできました。
https://github.com/euslisp/jskeus/blob/master/irteus/irtviewer.l#L283
でえらになっているので、bounding-box に:worldcoordsをつければいいだけ？

closed 2016-11-09 by h-kamada

chiwunau, 2016-11-09:
Bounding box -> eus のcubeなら昔Tabletopのsceneでやったことがあります。
必要であれば
https://github.com/chiwunau/seperate_tabletop_cluster/blob/master/euslisp/generate-tabletop-scene.l#L76-L118