euslisp/jskeus #401: kinematics viewer が上がってるとcoordsがdraw-onされない

YuOhara, 2016-12-01:
kinematics viewer ありだと
(send (make-coords) :draw-on :flush t) 
が上手く動かない気がします。

```
(load "package://hrpsys_ros_bridge_tutorials/euslisp/samplerobot-interface.l")
;; (SampleRobot)
(make-irtviewer)
(send (make-coords) :draw-on :flush t) ;; able to see
(read-line)
(samplerobot-init)
;; (objects *sr*)
(send (make-coords) :draw-on :flush t) ;; not able to see
```

mmurooka, 2016-12-01:
`(send (make-coords) :draw-on :flush t :viewer (send *irtviewer* :view))`
ではどうだろう．

YuOhara, 2016-12-01:
やっぱり見えないです・・
```
(send (make-coords) :draw-on :flush t :viewer (send *irtviewer* :viewer))
```

k-okada, 2016-12-02:
please try https://github.com/euslisp/jskeus/pull/402

YuOhara, 2016-12-02:
Thanks. It worked.

k-okada, 2016-12-03:
draw-on 系はよく使っているかな？
ちょっと使ってみて、副作用がないか教えてもらえると嬉しいです．

メモ：
各 draw-on で:flush t のときだけ、viewerをmake-currentするようにした．なので、
send a :draw-on
send b :draw-on
send c :draw-on :flush t
はだめなはず．やっぱ全ての場合でmake-currentするべきか．

YuOhara, 2016-12-03:
はい。私はcascoordsを新しく作ってdraw-onするなど使っていました。
しばらく使ってみます。

PR #402 (2016-12-30): add :makecurrent in :draw-on, this will fix https://github.com/euslisp/jskeus/issues/401
Close #401
closed 2016-12-30 by k-okada