euslisp/jskeus #403: Homebrew install error: SHA256 mismatch

knorth55, 2016-12-04:
OS X 10.12
Xcode 8.1

I tried to install `jskeus` via `homebrew` and got error below.

```
$ brew install homebrew/x11/jskeus
==> Installing jskeus from homebrew/x11
==> Downloading https://github.com/euslisp/jskeus/archive/1.0.13.tar.gz
==> Downloading from https://codeload.github.com/euslisp/jskeus/tar.gz/1.0.13
######################################################################## 100.0%
Error: SHA256 mismatch
Expected: 234ae02f97975ee5e38bd2c87921dd541c8ad1d52e4971321307a96f038773cf
Actual: 86437c939093d5c77776a6acf52453c49bdcb2b9d3b6e7956403ec34c476df7d
Archive: /Users/shingo/Library/Caches/Homebrew/jskeus-1.0.13.tar.gz
To retry an incomplete download, remove the file above.
```

wkentaro, 2016-12-05:
Maybe the tag is force pushed?

wkentaro, 2016-12-05:
Will be fixed via https://github.com/Homebrew/homebrew-x11/issues/275

wkentaro, 2016-12-11:
Now it is fixed. 😄 
[image]

closed 2016-12-12 by knorth55