euslisp/jskeus #415: centroid-on-footに類するものはありますか

k-okada, 2017-02-08:
昔, `send *robot* :centroid-on-foot` みたいな関数があった気がしたんですが、今はどうやってみなさん計算していますか？

@snozawa @mmurooka @yutakojio, @rkoyama1623

mmurooka, 2017-02-08:
:move-centroid-on-foot
のことでしょうか．

rkoyama1623, 2017-02-08:
https://github.com/euslisp/jskeus/issues/320
の流れだとすると、
:centroidを自分で投影したり、:calc-static-balance-pointを使ったりしています。

k-okada, 2017-02-08:
ああ、https://github.com/euslisp/jskeus/issues/320#issuecomment-274314954
とおなじ流れです．
:calc-static-balance-pointというのもあるのか．

```
3.irteusgl$ setq a (instance sample-robot :init)
#<sample-robot #X652d628 sample-robot  0.0 0.0 0.0 / 0.0 0.0 0.0>
4.irteusgl$ send a とい^Cinterrupt4.B1-irteusgl$
nil
4.B1-irteusgl$ send a :calc-static-balance-point
#f(1.2263 -1.399680e-15 -712.5)
5.B1-irteusgl$ send a :calc-zmp
#f(1.2263 -1.090750e-15 0.0)
6.B1-irteusgl$ send a :calc-zmp
#f(1.2263 -1.090750e-15 0.0)
7.B1-irteusgl$ send a :calc-zmp
#f(1.2263 -1.090750e-15 0.0)
```
calc-zmpとcalc-static-balance-point は座標系が違うのかな．
https://github.com/euslisp/jskeus/blob/master/irteus/irtdyna.l#L908-L914
https://github.com/euslisp/jskeus/blob/977985a7bbaeade22d0de11eae0ebf35698e7202/irteus/irtrobot.l#L1013-L1016
説明があるといいですね．

support-polygon 計算してからの、centroidの投影は各自プログラムを書いているのかな．

snozawa, 2017-02-09:
遅くなりました。
euslibの時代の`:centroid-on-foot`は、
> support-polygon 計算してからの、centroidの投影は各自プログラムを書いているのかな．

のとおり、足平形状などから計算されるsupport-polygonにcentroidを投影する(eusの`:foot`メソッドをつかう）ものでした。
したがって、
https://github.com/euslisp/jskeus/issues/415#issuecomment-278304396
https://github.com/euslisp/jskeus/issues/415#issuecomment-278307643
のものとは用途が違うようです。

それで、おそらくこのPRのニーズとしては、「バランスチェック計算するメソッドはあるか」、ということだと思いますが、
後々「support-polygon 計算してからの、centroidの投影」することが
バランスチェックの観点でとくに物理的に意味を持たないことが分かったので、euslibから移行してきていません。
足平が水平面になってればOKですが、傾いたらとくに意味がなくなり、足平が水平面のときには`(setf (elt centroid 2) foot-polygon-z-pos)`的に一行かいたらOKそうになります。

また、ニーズが「バランスチェック計算するメソッドはあるか」であってる場合、
さらに具体的にどういう状況のプログラムとして使いたいかがわかると良いかもしれないです。
（足平が水平になってる状況、左右それぞれ足平が水平でなく互い違いになってる、手足での接触を考えてる、動的か静的か、...etc）

> calc-zmpとcalc-static-balance-point は座標系が違うのかな．

これらの説明がなかった気がしますね。
docを更新したほうが良いですね。

ちなみに、
「support-polygon 計算してからの、centroidの投影」
でなくて
「support-polygonを水平面に投影して、その水平面support-polygonにcentroidを投影」
するようなプログラムのほうが、まだちょっとだけ物理的意味をもつケースがあることになりますが、
そういえばeusでfaceクラスを何かに投影するメソッドとかってあるんでしたっけ。

k-okada, 2017-03-03:
> #415 (comment)

おそくなりました．
やりたかったのは最終的にはバランスチェックだけど、途中のVisualizationはしたい．
ということで、シンプルにdual-manip-ikで、動的でも静的でもいいので、:centroid-on-footを表示して
て、バランスをチェックしてアウトなら色をかえて表示する、みたいなイメージです．

ちなみに、

> 左右それぞれ足平が水平でなく互い違いになってる、手足での接触を考えてる

で、バランスチェックはどうやあってやるのかな．

snozawa, 2017-05-23:
（こちらすごく遅くなりました）
参考情報としまして、３次元で厳密な（静的）バランスをとるためのプログラムとして、
[README](https://github.com/jsk-ros-pkg/jsk_control/tree/master/eus_qp)
[コード](https://github.com/jsk-ros-pkg/jsk_control/blob/master/eus_qp/euslisp/cfr-cwc-calculation.l)
[サンプル](https://github.com/jsk-ros-pkg/jsk_control/blob/master/eus_qp/euslisp/test-cfr-cwc-calculation.l)
があります(外部の最適化、この場合qp計算のライブラリを使うので、eus_qpの場所においてます）。
```
roseus
(load "package://eus_qp/euslisp/test-cfr-cwc-calculation.l")
(setup)
(demo-calc-cfr-6) ;; 1~7まであります
```
３次元的な接触でも、摩擦があっても、静的に維持可能な支持多角系を２Dのpolygonで計算してくれます