euslisp/jskeus #419: calc-zmpとcalc-static-balance-point は座標系が違うので説明をかく

k-okada, 2017-03-03:
https://github.com/euslisp/jskeus/issues/415#issuecomment-278515566

> これらの説明がなかった気がしますね。
> docを更新したほうが良いですね。

@rkoyama1623

rkoyama1623, 2017-03-04:
わかりました。

rkoyama1623, 2017-03-17:
`jskeus/doc/jmanual.pdf`には，docstringから生成されるドキュメントが記述されていますね．
calc-static-balance-pointもcalc-zmpも
docstringには，zmpの高さがキーワード引数で渡せることがすでに記述されていました（英語）．
https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L1016
https://github.com/euslisp/jskeus/blob/master/irteus/irtdyna.l#L913

ところが，`jskeus/doc/jmanual.pdf`には，
calc-static-balance-pointの記述はありますが，calc-zmpの記述はありません．
`robot-model`クラスの`calc-static-balance-point`メソッドは，
`irtrobot.l`で，クラスとメソッドの定義が同時に行われているのに対し，
`cascaded-link`クラスの`calc-zmp`メソッドは，クラスの定義がirtmodel.lにあり，メソッドの定義が`irtdyna.l`にあるため，`irtdyna.l`からドキュメントを自動生成する際に，うまくドキュメントが生成されていないようです．

解決策としては，`irtdyna.l`内で定義されている`cascaded-link`のメソッドを，`irtmodel.l`に移すと，
ドキュメントの自動生成が容易になるのと，
メソッドの定義が一箇所に集まるという意味でメンテナンス性も良いように思いますが，
ひょっとすると後方互換性等の問題もあるかもしれません．

k-okada, 2017-03-18:
> cascaded-linkクラスのcalc-zmpメソッドは，クラスの定義がirtmodel.lにあり，メソッドの定義がirtdyna.lにあるため，irtdyna.lからドキュメントを自動生成する際に，うまくドキュメントが生成されていないようです．

なるほど．問題があれば、それを解決するのがいいですね．ということで、これを直しました． > https://github.com/euslisp/EusLisp/issues/238