euslisp/jskeus #422: jskeusのjmanualのbuild方法について

rkoyama1623, 2017-03-15:
#419 に関連して，ドキュメントの生成方法について質問です．

手元のPCでeuslispとjskeusを最新にして，ドキュメント生成を試みると
```bash
$ cd jskeus/doc
mkdir -p /tmp/user/1000/tmp.7DcmXn2U9J/fig
cp ../eus/doc/jlatex/*.tex /tmp/user/1000/tmp.7DcmXn2U9J
cp: `../eus/doc/jlatex/*.tex' を stat できません: そのようなファイルやディレクトリはありません
make: *** [copy_eus_tex] エラー 1
```
となってしまいます．
どこかのバージョンまでは，`jskeus/eus`というeuslispのコピーと思われるディレクトリが自動で生成（おそらくcatkin build 実行時に）されていたと思うのですが，現在は生成されないためbuildに失敗しているように見えます．

jskeusのドキュメントの生成方法は現在，どのようになっているでしょうか？
#419 に関連して，ドキュメントの生成方法について質問です．

手元のPCでeuslispとjskeusを最新にして，ドキュメント生成を試みると
```bash
$ cd jskeus/doc
mkdir -p /tmp/user/1000/tmp.7DcmXn2U9J/fig
cp ../eus/doc/jlatex/*.tex /tmp/user/1000/tmp.7DcmXn2U9J
cp: `../eus/doc/jlatex/*.tex' を stat できません: そのようなファイルやディレクトリはありません
make: *** [copy_eus_tex] エラー 1
```
となってしまいます．
どこかのバージョンまでは，`jskeus/eus`というeuslispのコピーと思われるディレクトリが（おそらくcatkin build 実行時に）自動で生成されていたと思うのですが，現在は生成されないためbuildに失敗しているように見えます．

jskeusのドキュメントの生成方法は現在，どのようになっているでしょうか？

k-okada, 2017-03-16:
あ、これはcatkinはつかわず
```
$ hub clone euslisp/jskeus
$ cd jskeus/
$ make
$ make doc
```
とすることが想定されていました．

rkoyama1623, 2017-03-17:
教えていただいた方法でbuildできました．
ありがとうございました．

closed 2017-03-17 by rkoyama1623