euslisp/jskeus #434: Both move-target and link-list are needed for inverse-kinematics

pazeshun, 2017-06-06:
According to https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L2226, both move-target and link-list are needed for inverse-kinematics.
But error message in https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L2227 says either move-target or link-list is needed.
Which is correct?

k-okada, 2017-06-06:
Can you provide program to reproduce your problem?

k-okada, 2017-06-08:
I see, the error message is not correct, fixed in https://github.com/euslisp/jskeus/pull/435

but that is the `cascaded-link` level, usually you call :inverse-kinematics from `robot-model` https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L135-L145

and in this case, if you set `:move-target`, you do not need to specify `:link-list`, but if you set `:link-list` you have to specify `:move-target`

```
(load "irteus/demo/sample-robot-model.l")
(setq *robot* (instance sample-robot :init))
(objects (list *robot*))
;;
(warning-message 2 ":larm :inverse-kinematics~%")
(send *robot* :reset-pose)
(send *robot* :larm :inverse-kinematics (make-coords :pos #f(400 200 0)) :debug-view :no-message)
;;
(warning-message 2 ":inverse-kinematics :move-target~%")
(send *robot* :reset-pose)
(send *robot* :inverse-kinematics (make-coords :pos #f(400 200 0)) :move-target (send *robot* :larm :end-coords) :debug-view :no-message)

(warning-message 2 ":inverse-kinematics :link-list~%")
(send *robot* :reset-pose)
(send *robot* :inverse-kinematics (make-coords :pos #f(400 200 0)) :link-list (send *robot* :link-list (send *robot* :larm :end-coords :parent) (send *robot* :lleg :end-coords :parent)) :debug-view :no-message)
;; Error becasue of https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L135-L145
;(load "~/catkin_ws/ws_euslisp/src/jskeus/irteus/irtmodel.l")
;(load "~/catkin_ws/ws_euslisp/src/jskeus/irteus/irtrobot.l")
(send *robot* :reset-pose)
;(send-message *robot* cascaded-link :inverse-kinematics (make-coords :pos #f(400 200 0)) :link-list (send *robot* :link-list (send *robot* :larm :end-coords :parent) (send *robot* :lleg :end-coords :parent)) :debug-view :no-message)
;;(send-message *robot* cascaded-link :inverse-kinematics (make-coords :pos #f(400 200 0)) :move-target (send *robot* :larm :end-coords) :link-list (send *robot* :link-list (send *robot* :larm :end-coords :parent) (send *robot* :lleg :end-coords :parent)) :debug-view :no-message)
(send *robot* :inverse-kinematics (make-coords :pos #f(400 200 0)) :move-target (send *robot* :larm :end-coords) :link-list (send *robot* :link-list (send *robot* :larm :end-coords :parent) (send *robot* :rleg :end-coords :parent)) :debug-view :no-message)
```

pazeshun, 2017-06-08:
Sorry, I forgot to tell my debug.
Yes, you are right.

But, error occurs in jsk_apc when only `:move-taget` is specified:
```
1.irteusgl$ load "package://jsk_2016_01_baxter_apc/euslisp/jsk_2016_01_baxter_apc/baxter.l"
t
2.irteusgl$ setq *baxter* (instance jsk_2016_01_baxter_apc::baxter-robot :init)
#<jsk_2016_01_baxter_apc::baxter-robot #X58ac0f8 baxter  0.0 0.0 0.0 / 0.0 0.0 0.0>
3.irteusgl$ send *baxter* :inverse-kinematics (make-coords :pos #f(400 200 0)) :move-target (send *baxter* :larm :end-coords)
;; ERROR: :link-list or :move-target required
t
```
The cause of this is overwritten `:inverse-kinematics` in `package://jsk_2016_01_baxter_apc/euslisp/jsk_2016_01_baxter_apc/baxter.l`.
When `:link-list` is not specified in https://github.com/start-jsk/jsk_apc/blob/master/jsk_2016_01_baxter_apc/euslisp/jsk_2016_01_baxter_apc/baxter.l#L51, `:link-list nil` is passed in https://github.com/start-jsk/jsk_apc/blob/master/jsk_2016_01_baxter_apc/euslisp/jsk_2016_01_baxter_apc/baxter.l#L59.
If `:link-list nil` in https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L328, `link-list` is not set by https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L329-L330.

I think I should fix jsk_apc not to pass `:link-list nil` when `:link-list` is not specified, so I will make PR to jsk_apc.

k-okada, 2017-06-08:
it seems your question was solved. closing.

PR #435 (2018-07-11): cascaded-link: inverse-kinematics, fix error message both :link-list and :move-target is required
Close #434
closed 2017-06-08 by k-okada