euslisp/jskeus #436: copy-objectの使い方

708yamaguchi, 2017-06-08:
以下のようなプログラムをroseusで実行したのですが、IRTviewerに何も表示されません。
緑色の立方体が表示されると思ったのですが、copy-objectを使うとどうしておかしくなるのでしょうか。
```
(setq a (make-cube 100 100 100))
(send a :set-color :green)
(objects (list a))
(setq b (copy-object a))
(send b :set-color :red)
(objects (list a))
```

k-okada, 2017-06-08:
いい質問ですね
```
(setq a (make-cube 100 100 100))
(send a :set-color :green)
(setq b (copy-object a))
(send b :set-color :red)
(send b :locate #f(100 0 0))
(objects (list a b))
```
だと動きますよね．なので、
```
(setq a (make-cube 100 100 100))
(send a :set-color :green)
(objects (list a))
(setq b (copy-object a))
(setf (get b :gl-displaylist-id) nil)
(send b :set-color :red)
(objects (list a))
```
としても動きます．
https://github.com/euslisp/jskeus/blob/1.1.0/irteus/irtgl.l#L975-L976
がヒントで、
https://www.khronos.org/registry/OpenGL-Refpages/gl2.1/xhtml/glCallList.xml
http://www.hiramine.com/programming/modelviewer/09_displaylist.html
にあるようにGLのコードを呼ぶのではなく、メモリに蓄えてidで一気に呼び出すという高速化の方法です @ishiguroJSK

最初の`(object (list a))` でgllist が作られるのですが、その値が`copy-object`時にコピーされているのがおかしくなっているんだともいます．
gllistはobjectsを呼ば無いと出来ないため、最初の例はcopy-objectするまでobjectsを呼ばない方法．
二個目の例は出来てしまったgllistを消しています．

708yamaguchi, 2017-06-09:
ありがとうございます。

以下のプログラムを実行したところ、
```
(setq a	(make-cube 100 100 100))
(objects (list a))
(setq b	(copy-object a))
(print (get a :gl-displaylist-id))
(print (get b :gl-displaylist-id))
```
次のような結果になり、
```
((88767520 . 1))
((88767520 . 1))
```
確かに`:gl-displaylist-id`の値がコピーされていることが確認できました。