euslisp/jskeus #455: make-prismで作った立体におけるpqp-collision-checkのバグ

chibi314, 2017-08-28:
図にあるような2つの物体がpqp-collision-checkを行うと干渉するというバグがありました。
[image]

T字の物体はmake-prismを使って作ったもの、もうひとつの物体はmake-cubeで作った細長い直方体です。
#454 と関連するのですが、面の方向を逆にしても干渉判定が出ました。また、面の方向を逆にしてさらにT字の厚みを増すと干渉していないと判定されました。以下、サンプルコードです。

```
(defun test-pqp-0 ()
  "正しくない"
  (setq obj1 (make-prism (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0)) 10))
  (setq obj2 (make-cube 1.0 1.0 100))
  (send obj2 :newcoords (make-coords :pos #f(-50 50 0)))
  (objects (list obj1 obj2))
  (print (pqp-collision-check obj1 obj2))
  )

(defun test-pqp-1 ()
  "正しい（面を反転、Tの厚みは100）"
  (setq obj1 (make-prism (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0)) 100))
  (dolist (f (send obj1 :faces))
    (send f :invert))
  (setq obj2 (make-cube 1.0 1.0 100))
  (send obj2 :newcoords (make-coords :pos #f(-50 50 0)))
  (objects (list obj1 obj2))
  (print (pqp-collision-check obj1 obj2))
  )

(defun test-pqp-2 ()
  "正しくない（面を反転、Tの厚みは10）"
  (setq obj1 (make-prism (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0)) 10))
  (dolist (f (send obj1 :faces))
    (send f :invert))
  (setq obj2 (make-cube 1.0 1.0 100))
  (send obj2 :newcoords (make-coords :pos #f(-50 50 0)))
  (objects (list obj1 obj2))
  (print (pqp-collision-check obj1 obj2))
  )

```

YoheiKakiuchi, 2017-08-29:
https://github.com/euslisp/jskeus/issues/454
ここで出ている、正しい法線方向のprismを作ったらどうなりますか？

また、pqp-collision-checkではなくて、pqp-collision-distanceで (list distance p1 p2) が返ってくるのですが、そのときのp1とp2はどこになっていますか？

s-noda, 2017-08-29:
見当違いのことを言っているかもしれませんが，
make-prism は閉じないといけないのではないでしたっけ？
以下のように make-prism (list #F(hoge) ... #F(hoge)) とするとすべて pqp-collision-check = 0 となるようですが．

(defun test-pqp-0 ()
  "正しくない"
  (setq obj1 (make-prism (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 \
0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0) #F(15 0 0)) 10))
  (setq obj2 (make-cube 1.0 1.0 100))
  (send obj2 :newcoords (make-coords :pos #f(-50 50 0)))
  (objects (list obj1 obj2))
  (print (pqp-collision-check obj1 obj2))
  )

(defun test-pqp-1 ()
  "正しい（面を反転、Tの厚みは100）"
  (setq obj1 (make-prism (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 \
0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0) #F(15 0 0)) 100))
  (dolist (f (send obj1 :faces))
    (send f :invert))
  (setq obj2 (make-cube 1.0 1.0 100))
  (send obj2 :newcoords (make-coords :pos #f(-50 50 0)))
  (objects (list obj1 obj2))
  (print (pqp-collision-check obj1 obj2))
  )

(defun test-pqp-2 ()
  "正しくない（面を反転、Tの厚みは10）"
  (setq obj1 (make-prism (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 \
0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0) #F(15 0 0)) 10))
  (dolist (f (send obj1 :faces))
    (send f :invert))
  (setq obj2 (make-cube 1.0 1.0 100))
  (send obj2 :newcoords (make-coords :pos #f(-50 50 0)))
  (objects (list obj1 obj2))
  (print (pqp-collision-check obj1 obj2))
  )

YoheiKakiuchi, 2017-08-29:
一応、マニュアル通りの以下で6面体ができているようだ。
```
(make-prism (list #f(100 100 0) #f(100 -100 0) #f(-100 -100 0) #f(-100 100 0)) 200)
```
閉じるようにすると、7面体になって、面積0 (eus的にはnanが返ってくる)の面ができる
```
(make-prism (list #f(100 100 0) #f(100 -100 0) #f(-100 -100 0) #f(-100 100 0) #f(100 100 0)) 200)
```

s-noda, 2017-08-29:
となると pqp だけの話ですかね．

```
(defun test-pqp-loop (&optional (close? t)
                                (p (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) \
#f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0))))
  (setq obj1 (make-prism (append p (if close? (list (copy-seq (car p))))) 10))
  (setq obj2 (make-cube 1.0 1.0 100))
  (objects (list obj1 obj2))
  (let* ((cnt 0) col)
    (do-until-key
     (send obj2 :newcoords (make-coords :pos (float-vector (* 50 (sin (* 0.01 (incf cnt))))\
 (* 50 (cos (* 0.01 cnt))) 0)))
     (send *viewer* :draw-objects)
     (x::window-main-one)
     (setq col (print (pqp-collision-check obj1 obj2)))
     (if (eq col 1) (send obj1 :set-color (float-vector 1 0 0))
       (send obj1 :set-color (float-vector 0 1 0)))
     (unix::usleep (* 10 1000)))))
```

ぶつかるとあかくなります．
test-pqp-loop t は正しそう．
[image]

test-pqp-loop nil は へこんでる部分がつながって衝突判定されているとかでしょうか
[image]

k-okada, 2017-08-29:
つまり，pqpに渡すときに閉じる必要があるということ？ PR＋テストコード作れるかな？

mmurooka, 2017-08-29:
pqp.lを見てみて
https://github.com/euslisp/jskeus/blob/master/irteus/pqp.l#L40
の`face-to-triangle-aux`の返り値がおかしいようです．
```
(defun test-pqp-0-draw-face-triangle
  ()
  (setq obj1 (make-prism (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0)) 10))
  (objects (list obj1))
  (send-all (geometry::face-to-triangle (elt (send obj1 :faces) 0)) :draw-on :flush t)
  )
```
で添付画像のように，物体の外の領域を埋めるような三角形ができてしまっています．
[image]

mmurooka, 2017-08-29:
```
(defun test-face-to-triangle-with-close-poly
  ()
  (unless (boundp '*irtviewer*)
    (make-irtviewer))
  (setq poly1 (make-polygon #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0) #f(15 0 0)))
  (send-all (geometry::face-to-triangle poly1) :draw-on :flush t)
  )

(defun test-face-to-triangle-with-open-poly
  ()
  (unless (boundp '*irtviewer*)
    (make-irtviewer))
  (setq poly1 (make-polygon #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0)))
  (send-all (geometry::face-to-triangle poly1) :draw-on :flush t)
  )
```

`(test-face-to-triangle-with-close-poly)`の結果
[image]

`(test-face-to-triangle-with-open-poly)`の結果
[image]

YoheiKakiuchi, 2017-08-29:
どこかにバグがあるような気もするけど、マニュアル通りの記法で正しい面の向き（外側に法線が向く）にすると、問題ないように見えるけれども。

~~~
(defun test-pqp-0-draw-face-triangle ()
  (setq obj1 (make-prism (reverse (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0\
 0))) 10))
  (objects (list obj1))
  (send-all (geometry::face-to-triangle (elt (send obj1 :faces) 0)) :draw-on :flush t)
  )
~~~

[image]

mmurooka, 2017-08-29:
そのときは`(elt (send obj1 :faces) 0)`を`(elt (send obj1 :faces) 1)`として，上の面にすると，ダメな三角形が表示されますね．

chibi314, 2017-08-29:
皆さんありがとうございます。
一つ報告を忘れていることがありました。

```
(make-prism (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0)) 10)
```
を

```
(make-prism (list #f(15 0.1 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0)) 10)
```
とすると結果が正しくなるということが分かっていました。Tの上の棒が一直線になっていることが何かバグの原因でしょうか？以下、結果です。

```
(defun test-pqp-0-draw-face-triangle
    ()
  (setq obj1 (make-prism (list #f(15 0.1 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0)) 10))
  (objects (list obj1))
  (send-all (geometry::face-to-triangle (elt (send obj1 :faces) 0)) :draw-on :flush t)
    )

(defun test-pqp-1-draw-face-triangle
    ()
  (setq obj1 (make-prism (list #f(15 0.1 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0)) 10))
  (objects (list obj1))
  (send-all (geometry::face-to-triangle (elt (send obj1 :faces) 1)) :draw-on :flush t)
    )

test-pqp-0-draw-face-triangleの結果
![result0](https://user-images.githubusercontent.com/7079117/29812332-28afc4ca-8ce1-11e7-96c1-314d2a8d37db.png)
test-pqp-1-draw-face-triangleの結果
![result1](https://user-images.githubusercontent.com/7079117/29812346-34bb013a-8ce1-11e7-9ac8-d1e31944c90a.png)

YoheiKakiuchi, 2017-08-30:
> とすると結果が正しくなるということが分かっていました。Tの上の棒が一直線になっていることが何かバグの原因でしょうか？以下、結果です。

そうですね。現状では、エッジを伸ばしていった先にverticesがあると、このようになることがあるコードになっているように見えます。

簡単なテストコードは、こんな感じ
~~~
(setq f (instance face :init :vertices (list #f(15 0 0) #f(15 120 0) #f(-15 120 0) #f(-15 0 0) #f(-120 0 0) #f(-120 -30 0) #f(120 -30 0) #f(120 0 0))))

;; debug draw
(objects (list (make-cube 300 300 1)))
(send f :draw-on :flush t :color #f(1 0 0))
(send-all (geo::face-to-triangle f) :draw-on :flush t :color #f(0 1 0) :width 2)

;;
(eq :outside (send f :insidep #f(-50 50 0))) ;; t
(eq :outside (send f :insidep #f(50 50 0))) ;; t
;;
(every #'(lambda (x) (eq :outside x)) (send-all (geo::face-to-triangle f) :insidep #f(-50 50 0))) ;; nil , failure, it should return t
(every #'(lambda (x) (eq :outside x)) (send-all (geo::face-to-triangle f) :insidep #f(50 50 0))) ;; t
;;
(eq :inside (send f :insidep #f(0 10 0))) ;; t
(some #'(lambda (x) (eq :inside x)) (send-all (geo::face-to-triangle f) :insidep #f(0 10 0))) ;; t
~~~