euslisp/jskeus #463: ik解くときに:debug-view tしても早すぎて何が起こっているかわからない

k-okada, 2017-11-01:
https://github.com/euslisp/jskeus/blob/1.1.0/irteus/irtmodel.l#L1988
の
`:additional-check `
というのを見つけたのですが，
```
       (if additional-check
           (setq success (and success (funcall additional-check))))
```
なので，IKが成功したときにhookがかかるみたいです．IKの毎ループでhookをかける方法があれば，知りたいです．
https://github.com/euslisp/jskeus/blob/1.1.0/irteus/irtmodel.l#L2066-L2067

mmurooka, 2017-11-01:
オフラインで話していましたが，以下だとできます．もっと綺麗な方法をご存じの方教えてください．
```
(load "package://jskeus/irteus/demo/sample-robot-model.l") ;; load installed robot
(setq *robot* (instance sample-robot :init))
(objects (list *robot*))
(send *robot* :init-pose)
(send *robot* :rarm :inverse-kinematics
      (list #'(lambda () (unix::usleep (* 100 1000)) (make-coords :pos (float-vector 1e3 0 0))))
      :debug-view :no-message
      :stop 100)
```

iory, 2017-11-01:
僕が発見したのはmove-joints-hook

k-okada, 2017-11-02:
```
            (send *pr2* :larm :inverse-kinematics (make-coords :pos #f(800 0 800))
              :rotation-axis :z :debug-view t :p-limit 10 :r-limit 0.05)
```
というのを  @dyson8910 と見つけました．

iory, 2017-11-02:
p-limitとr-limitで１stepで移動する関節の速度を制限することで，ikのiteration回数が増えるのでdebug-viewした時にゆっくりに動いて見えるということですか．https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L2516-L2550

分かりやすい真っ当な方法な気がします