euslisp/jskeus #464: :move-end-posの引数に:localがないとエラーになる

708yamaguchi, 2017-11-08:
@k-okada 

fetchで`move-end-pos`を`:use-torso nil`で使いたかったのですが、
```lisp
84.irteusgl$ (send *fetch* :rarm :move-end-pos #f(0 -100 0) :use-torso nil)
:parent-orientation wrt?err:
```
のようになりました。

`irtrobot.l`を読んで`wrt`というものが必要らしく
https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L161-L165
```lisp
86.irteusgl$ (send *fetch* :rarm :move-end-pos #f(0 -100 0) :local :use-torso nil)
#f(203.985 38.6729 86.9734 -106.043 100.21 -107.811 37.269 -71.58 0.181272 1.17748)
```
としたら所望の動きになりました。

この`wrt`は`:local`以外の使い方があるんでしょうか？
また`wrt`がない状態（最初にエラーが出たコード）で動かせないでしょうか？

708yamaguchi, 2017-11-08:
cc @knorth55

knorth55, 2017-11-09:
もしも`wrt`の候補が列挙可能なのであれば
```
(when
  (find (car args) (list :world :local :parent))
  (setq wrt (pop args))
```
という風に回避できそうですが列挙可能なのでしょうか？

knorth55, 2017-11-09:
jmanualをみると

wrt は :local, :parent, :world のキーワードあるいは coordinates のインスタンスでなければな
らない。

とかかれているのでこれをチェックするように変更すればいいのでしょうか

k-okada, 2017-11-13:
https://github.com/euslisp/EusLisp/blob/EusLisp-9.23/lisp/l/coordinates.l#L225 のエラーメッセージを追加(wrt must be either :world, :local or :parent) したらいいのかな，という気もするけどダメかな @708yamaguchi

708yamaguchi, 2017-11-13:
そうですね。`wrt`を指定しなくても使えるようにするのではなくて、ちゃんと指定させるようにエラーメッセージを出すのが正しい気がしてきました。
pull request作ります。

708yamaguchi, 2017-11-13:
https://github.com/euslisp/EusLisp/pull/252

knorth55, 2017-11-13:
#465 はcloseしますか？
それともこちらも追加しますか？

knorth55, 2017-11-27:
#465 をcloseしました