euslisp/jskeus #466: Minjerk-interpolator's last position value differs from goal

Affonso-Gui, 2017-11-17:
When using large (over 100?) time values for minjerk-interpolator, calculation error causes the final position to differ from goal:
```
(setq l (instance minjerk-interpolator :init))
(send l :reset :position-list (list #f(1 2 3) #f(3 4 5) #f(1 2 3)) :time-list (list 1000 1800))
(send l :start-interpolation)
(while (send l :interpolatingp) (send l :pass-time 200) (print (send l :position)))
(send l :position)                ;; #f(1.0 2.0 3.0)                                                  
(v= #f(1 2 3) (send l :position)) ;; nil                                                              
(v- #f(1 2 3) (send l :position)) ;; #f(1.421085e-14 1.421085e-14 1.421085e-14)                       
```
Became a problem in [robot-interface's end-coords-interpolation](https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L475), since `p` never reaches 1.0 due to the calculation error.

Problem is avoided by using smaller time values, as `:interpolation` documentation tells us:
```
#<metaclass #X5dcd7d8 minjerk-interpolator>
Minjerk interpolator, a.k.a Hoff & Arbib

    Example code is:
(setq l (instance minjerk-interpolator :init))
(send l :reset :position-list (list #f(1 2 3) #f(3 4 5) #f(1 2 3)) :time-list (list 0.1 0.18))
(send l :start-interpolation)
(while (send l :interpolatingp) (send l :pass-time 0.02) (print (send l :position)))
``` 
However, actual implementations seems to use bigger values (probably using in ms), as we can see with:
```
(send *ri* :angle-vector (send *pr2* :reset-pose))
(v= (send *ri* :reference-vector) (send *pr2* :angle-vector)) ;; nil                                  
(v- (send *ri* :reference-vector) (send *pr2* :angle-vector)) ;; #f(-2.842171e-14 -2.842171e-14 -5.68\
4342e-14 -5.684342e-14 5.684342e-14 -2.842171e-14 1.421085e-14 -2.273737e-13 2.842171e-14 -5.684342e-\
14 5.684342e-14 5.684342e-14 2.842171e-14 5.684342e-14 -3.410605e-13 0.0 0.0)
```
This could also be solved by returning actual goal instead of calculated position in [irtutil.l](https://github.com/euslisp/jskeus/blob/master/irteus/irtutil.l#L281).

My question is:  **Should we fix minjerk-interpolator to ignore calculation error or simply be careful when choosing interpolation time?**

k-okada, 2017-11-20:
> This could also be solved by returning actual goal instead of calculated
position in irtutil.l.

please create PR for this, because min-jerk might be used in motion
generation people, so we need more discussion.

to check the equality between `(send l :position)` and `#f(1 2 3)`, you
should use `eps-v=` instead of `v=`, something like
```
(print (eps-v= #f(1 2 3) (send l :position) 0.001)) ;; nil

``

--
◉ Kei Okada

2017-11-17 17:31 GMT+09:00 Affonso, Guilherme <notifications@github.com>:

> When using large (over 100?) time values for minjerk-interpolator,
> calculation error causes the final position to differ from goal:
>
> (setq l (instance minjerk-interpolator :init))
> (send l :reset :position-list (list #f(1 2 3) #f(3 4 5) #f(1 2 3)) :time-list (list 1000 1800))
> (send l :start-interpolation)
> (while (send l :interpolatingp) (send l :pass-time 200) (print (send l :position)))
> (send l :position)                ;; #f(1.0 2.0 3.0)
> (v= #f(1 2 3) (send l :position)) ;; nil
> (v- #f(1 2 3) (send l :position)) ;; #f(1.421085e-14 1.421085e-14 1.421085e-14)
>
> Became a problem in robot-interface's end-coords-interpolation
> <https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L475>,
> since p never reaches 1.0 due to the calculation error.
>
> Problem is avoided by using smaller time values, as :interpolation
> documentation tells us:
>
> #<metaclass #X5dcd7d8 minjerk-interpolator>
> Minjerk interpolator, a.k.a Hoff & Arbib
>
>     Example code is:
> (setq l (instance minjerk-interpolator :init))
> (send l :reset :position-list (list #f(1 2 3) #f(3 4 5) #f(1 2 3)) :time-list (list 0.1 0.18))
> (send l :start-interpolation)
> (while (send l :interpolatingp) (send l :pass-time 0.02) (print (send l :position)))
>
> However, actual implementations seems to use bigger values (probably using
> in ms), as we can see with:
>
> (send *ri* :angle-vector (send *pr2* :reset-pose))
> (v= (send *ri* :reference-vector) (send *pr2* :angle-vector)) ;; nil
> (v- (send *ri* :reference-vector) (send *pr2* :angle-vector)) ;; #f(-2.842171e-14 -2.842171e-14 -5.68\
> 4342e-14 -5.684342e-14 5.684342e-14 -2.842171e-14 1.421085e-14 -2.273737e-13 2.842171e-14 -5.684342e-\
> 14 5.684342e-14 5.684342e-14 2.842171e-14 5.684342e-14 -3.410605e-13 0.0 0.0)
>
> This could also be solved by returning actual goal instead of calculated
> position in irtutil.l
> <https://github.com/euslisp/jskeus/blob/master/irteus/irtutil.l#L281>.
>
> My question is: *Should we fix minjerk-interpolator to ignore calculation
> error or simply be careful when choosing interpolation time?*
>
> —
> You are receiving this because you are subscribed to this thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/euslisp/jskeus/issues/466>, or mute the thread
> <https://github.com/notifications/unsubscribe-auth/AAeG3KFOI60JobnKt4r5lpux7pVIEEb-ks5s3URygaJpZM4Qhstq>
> .
>

Affonso-Gui, 2017-11-20:
PR created at #469 .

Let's discuss more there, and thanks for teaching about `eps-v=` , I did not know about this function.

closed 2018-09-03 by Affonso-Gui