euslisp/jskeus #468: [jmanual] :calc-torque

M-Onitsuka, 2017-11-17:
重力補償トルクを得るrobot-modelのメソッドとして
```:calc-torque```
がjmanualに載っていなかったので報告しておきます

k-okada, 2017-11-20:
関数定義の引数の後にコメントを入れるとマニュアルに乗るはずなので，追加しておいてください．
例：https://github.com/euslisp/jskeus/blob/master/irteus/irtrobot.l#L395

future731, 2019-05-07:
以下のdiffのように変更した上で，.gitのある場所でmakeを行ったところ，
jmanual, manualともに:calc-torqueがdocumentされなかったのですが，これはどうデバッグするのがよいでしょうか

diff --git a/irteus/irtdyna.l b/irteus/irtdyna.l
index 34fd6d4..37ec86e 100644
--- a/irteus/irtdyna.l
+++ b/irteus/irtdyna.l
@@ -707,6 +707,7 @@
          (root-coords (send (car (send self :links)) :copy-worldcoords))
          (force-list) (moment-list) (target-coords)
          (calc-torque-buffer-args (send self :calc-torque-buffer-args)))
+   "return torque"
    ;; check length
    (unless (= (length force-list) (length moment-list) (length target-coords))
      (error ";; ERROR : list length differ in :calc-torque : force-list ~A moment-list ~A target-coords ~A~%"

また，
https://github.com/euslisp/jskeus/blob/master/irteus/irtdyna.l#L901
にある:calc-zmpはdocumentedなはずですが，同じくマニュアルに見つかりませんでした．

future731, 2019-05-07:
この問題は
https://github.com/euslisp/EusLisp/pull/238
https://github.com/euslisp/jskeus/issues/419
あたりで解決していそうに見えるので，何か私がコンパイル方法を間違えている可能性がある気がしました．

ubuntu16.04の環境において，適当なディレクトリで
```
git clone https://github.com/euslisp/jskeus.git
cd jskeus
make
xdg-open jmanual.pdf # or xdg-open manual.pdf
```
をしています．

aptのEusLispのバージョンを確認したところ，
```
$ dpkg -l | grep kinetic-euslisp
ii  ros-kinetic-euslisp                                         9.26.0-0xenial-20190320-081347-0800                   amd64        EusLisp is an integrated programming system for the research on intelligent robots based on Common Lisp and Object-Oriented programming
```

となっていて，9.26.0なのでコミット日時を考えると正しく動くべきもののようです．