euslisp/jskeus #473: let のスコープがおかしい

708yamaguchi, 2017-12-04:
```
(defun test1()
  (let ((a '(0 0)))
    (setf (elt a 0) (+ 1 (elt a 0)))
    a))
```
以上のように`test1`関数を定義して、それを２回以上呼び出すと、下のように返り値がおかしくなります。
```
1.irteusgl$ test1
(1 0)
2.irteusgl$ test1
(2 0)
3.irteusgl$ test1
(3 0)
```
`test1`関数の中でローカル変数`a`のスコープが終わると思うので、返り値は毎回`(1 0)`になるのが正しいのかなぁと思いました。

一方、`'(0 0)` の代わりに`(list 0 0)`を用いた以下の`test2`関数だと、
```
(defun test2()
  (let ((a (list 0 0)))
    (setf (elt a 0) (+ 1 (elt a 0)))
    a))
```
以下のようになり、予想通りの結果になりました。
```
17.irteusgl$ test2
(1 0)
18.irteusgl$ test2
(1 0)
19.irteusgl$ test2
(1 0)
```

Cc @ban-masa

Affonso-Gui, 2017-12-04:
Just for record, in this case `clisp` behaves as `euslisp`, and `sbcl` gives the warning
```
 warning: 
    Destructive function SB-KERNEL:%SETELT called on constant data.
```
and return `(1 0)` every time.

inabajsk, 2017-12-04:
なかなかいい例ですね．
setfでaにバインドされた定数をtest1が書き換えると
test1の定義本体の定数を変更していることになるので
(test1)
pf test1
とするとtest1の定義自体が
'(0 0)から，'(1 0)に代わっていっている
ということになってゆくという例になりますね．

lispは，関数の本体定義自体もリストデータで表現される
ので，ここでは，その関数定義全体の中にあるリストデータ
を書き換えていることになってゆくからということに
なりますね．

稲葉

708yamaguchi, 2017-12-04:
@inabajsk 
早速のご返信ありがとうございます。
前のコメントで定義した`test1`と`pf`を交互に実行したところ、以下のような結果になり、`test1`の定義が変わっていくことが確認できました。
```
2.irteusgl$ pf test1
(lambda nil (let ((a '(0 0))) (setf (elt a 0) (+ 1 (elt a 0))) a))
nil
3.irteusgl$ test1
(1 0)
4.irteusgl$ pf test1
(lambda nil (let ((a '(1 0))) (setf (elt a 0) (+ 1 (elt a 0))) a))
nil
5.irteusgl$ test1
(2 0)
6.irteusgl$ pf test1
(lambda nil (let ((a '(2 0))) (setf (elt a 0) (+ 1 (elt a 0))) a))
nil
7.irteusgl$ test1
(3 0)
```