euslisp/jskeus #474: irteusのカメラでロボットの色が青みがかっている

mmurooka, 2017-12-13:
```
(load "package://hrpsys_ros_bridge_tutorials/euslisp/hrp2jsknts-interface.l")
(load "package://pr2eus/pr2-interface.l")
(load "package://hrpsys_ros_bridge_tutorials/euslisp/jaxon_red-interface.l")
(load "package://hrpsys_ros_bridge_tutorials/euslisp/samplerobot-interface.l")

(setq *hrp2* (instance hrp2jsknts-robot :init))
(setq *pr2* (instance pr2-robot :init))
(setq *jaxon* (instance jaxon_red-robot :init))
(setq *sr* (instance samplerobot-robot :init))

(setq *cam* (make-camera-from-param :pwidth 640 :pheight 360 :fx 400 :fy 400 :cx 319.5 :cy 179.5 :name "camtest" :create-viewer t))
(send *cam* :newcoords (send (make-coords :pos #f(3000 0 1000) :rpy (list 0 -pi/2 0)) :rotate pi/2 :z))
(setq *cube* (make-cube 100 100 100 :pos #f(1000 0 1000)))
(send *cube* :set-color #f(1.0 0.0 0.0))

(send *hrp2* :newcoords (make-coords :pos #f(0 -500 0)))
(send *pr2* :newcoords (make-coords :pos #f(0 500 0)))
(send *jaxon* :newcoords (make-coords :pos #f(0 -1000 0)))
(send *sr* :newcoords (make-coords :pos #f(0 1000 0)))
(objects (list *cube* *hrp2* *pr2* *jaxon* *sr*))
(send *cam* :draw-objects (list *cube* *hrp2* *pr2* *jaxon* *sr*))
```

irtviewer
[image]

camera
[image]

cc @iory

iory, 2017-12-13:
Thank you.
I changed ambient parameter in "hrp2jsknts.l" and the camera-view became a little better.
Simply, I changed the value of ambient param to the same value as diffuse param.

[image]
[image]

mmurooka, 2017-12-13:
irtviewerが暗いのを直した時のIssueとPR
https://github.com/jsk-ros-pkg/jsk_model_tools/issues/59
https://github.com/euslisp/jskeus/pull/141

YoheiKakiuchi, 2017-12-13:
@mmurooka 
の推測があたっていそうですね。
カメラモデルにlightの設定がされていない。
irtviewerと同じ設定にすると良いと思う。

iory, 2017-12-13:
@YoheiKakiuchi 
これはカメラのオブジェクトの中で，
```
  (:gllight 
   (&key (light_ambient (float-vector 0.1 0.1 0.1 1.0))
         (light_diffuse (float-vector 1.0 1.0 1.0 1.0))
         (light_specular (float-vector 0.1 0.1 0.1 1.0))
         (light0_position (float-vector 4000.0 3000.0 2000.0 0.0))
         (light1_position (float-vector -4000.0 -2000.0 0.0 0.0))
         (light2_position (float-vector -2000.0 -2000.0 8000.0 0.0))
         (light1_scale 0.7) (light2_scale 0.4))
    (let ()
      (glLightfv GL_LIGHT0 GL_AMBIENT light_ambient)
      (glLightfv GL_LIGHT0 GL_DIFFUSE light_diffuse)
      (glLightfv GL_LIGHT0 GL_SPECULAR light_specular)
      (glLightfv GL_LIGHT0 GL_POSITION light0_position)

      (glLightfv GL_LIGHT1 GL_AMBIENT (scale light1_scale light_ambient))
      (glLightfv GL_LIGHT1 GL_DIFFUSE (scale light1_scale light_diffuse))
      (glLightfv GL_LIGHT1 GL_SPECULAR (scale light1_scale light_specular))
      (glLightfv GL_LIGHT1 GL_POSITION light1_position)

      (glLightfv GL_LIGHT2 GL_AMBIENT (scale light2_scale light_ambient))
      (glLightfv GL_LIGHT2 GL_DIFFUSE (scale light2_scale light_diffuse))
      (glLightfv GL_LIGHT2 GL_SPECULAR (scale light2_scale light_specular))
      (glLightfv GL_LIGHT2 GL_POSITION light2_position))
    )
```
というようなことを行えばいいということでしょうか？

YoheiKakiuchi, 2017-12-13:
そうですね。
こんな感じでirtviewerと同じ表示になってます。 https://github.com/euslisp/jskeus/pull/475
drawする毎に、lightの設定しなくても良いかなと思って、viewerを作った時だけ設定になってます。

iory, 2017-12-13:
@YoheiKakiuchi さん

PRのコードを呼んで試してみたら，
黒塗りの影絵のようになりました．
@mmurooka さんの環境でも再現しているようです．

[image]
[image]

YoheiKakiuchi, 2017-12-13:
すいません。
lightingのenableとdisableを逆にしてコミットしてました。
いま直ってると思います。

iory, 2017-12-13:
ありがとうございます．
[image]
[image]

snozawa, 2017-12-13:
（別issue向けで、また前からあった現象ですが）
gl verticesの一部の色が、別なbodyか何かの色（今だとHRP2の襟元などが赤い色）になってますね。

YoheiKakiuchi, 2017-12-13:
> gl verticesの一部の色が、別なbodyか何かの色（今だとHRP2の襟元などが赤い色）になってますね

この色が付いている部分は裏面ですね。
裏面は裏とわかるように、表と違う色を付けようと思ったのだけど、現状は意図したコードが入ってなくて、
直前に設定した色に引っ張られるようになってしまってるようですね。

iory, 2017-12-20:
メモです．

hrp2のdaeファイルをblenderで表示させると，色が正しくついていそうなので，
colladaからeusに変換するところで，ambientの値が落ちているようです．

[image]

iory, 2017-12-20:
と思いましたが，やっぱり，vrml -> colladaでambientが設定されていないようです．
https://github.com/fkanehiro/openhrp3/pull/133

YoheiKakiuchi, 2017-12-21:
これがvrmlの定義。
http://www.wakayama-u.ac.jp/~tokoi/vrml/reference.html#appearance
```
Material ノード
Material {
  ambientIntensity 環境光の反射率
  diffuseColor r g b
```

マージされちゃったけど、
```
ambient = diffuse * ambientIntensity
```
となるのが、定義どおりと思う。